nordicsemi/IOS-DFU-Library #258: DFU Error 302...

mickaelhero, 2018-12-31:
Hello, The update of the dru does not work, here is the output:

`Message: [Callback] Central Manager did update state to: Powered ON
Message: Connecting to SomyBand...
Message: centralManager.connect(peripheral, options: nil)
state:Connecting
Message: [Callback] Central Manager did connect peripheral
Message: Connected to SomyBand
Message: Discovering services...
Message: peripheral.discoverServices(nil)
Message: Services discovered
Message: Starting Secure DFU...
Message: Connected to SomyBand
Message: Services discovered
Message: Secure DFU Service found
Message: Discovering characteristics in DFU Service...
Message: peripheral.discoverCharacteristics(nil, for: FE59)
Message: DFU characteristics discovered
Message: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
state:Starting
Message: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
Message: Buttonless DFU indications enabled
Message: Application with buttonless update found
Message: Trying setting bootloader name to Dfu37021
Message: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.writeValue(0x02084466753337303231, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
state:Enabling DFU Mode
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected by the remote device
Message: Scanning for the DFU Bootloader...

Message: DFU Bootloader found
Message: Connecting to Unknown device...
Message: centralManager.connect(peripheral, options: nil)
Message: [Callback] Central Manager did connect peripheral
Message: Connected to Unknown device
Message: Discovering services...
Message: peripheral.discoverServices([FE59])
Message: Services discovered
Message: DFU Service not found
Message: No services found
Message: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
Message: Disconnecting...
Message: centralManager.cancelPeripheralConnection(peripheral)
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected
DFU Error: 302
`

Here my code:

`func startUpdateDFU(){
        guard let url = Defaults[.firmwareURL] else {
            print("No file found")
            return
        }
        
        let selectedFirmware = DFUFirmware(urlToZipFile: url)
        
        guard
            let firmware = selectedFirmware,
            let peripheral = BLEManager.shared.peripheralManager.peripheral else { return }
        
        let initiator = DFUServiceInitiator(queue: nil).with(firmware: firmware)
        initiator.logger = self
        initiator.delegate = self
        initiator.progressDelegate = self
        initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
        initiator.peripheralSelector = self
        
        dfuController = initiator.start(target: peripheral)
    }`

Can someone help me to know why it does not work thanks

philips77, 2018-12-31:
Hello,
Did you follow the hint from the log?
```
Message: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
```

mickaelhero, 2019-01-01:
Yes, I am well connected to the right device, and the service exists.
I made a text with the app do not connect and the device meets well today, so the problem can not come from the Bluetooth device ..

If we look at the log, he connected to the device SomyBand, and suddenly it becomes strange from there:

```
...
state:Enabling DFU Mode
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected by the remote device
Message: Scanning for the DFU Bootloader...

Message: DFU Bootloader found
Message: Connecting to Unknown device...
Message: centralManager.connect(peripheral, options: nil)
Message: [Callback] Central Manager did connect peripheral
Message: Connected to Unknown device
Message: Discovering services...
Message: peripheral.discoverServices([FE59])
Message: Services discovered
Message: DFU Service not found
Message: No services found
Message: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
Message: Disconnecting...
Message: centralManager.cancelPeripheralConnection(peripheral)
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected
```

philips77, 2019-01-01:
Does it have the Service Changed characteristic both in app and bootloader mode?

mickaelhero, 2019-01-01:
I think, because with the app nrf connect I managed to update the bluetooth device

erezhod, 2019-01-03:
Following! I am having the same error :(

philips77, 2019-01-07:
> 
> 
> I think, because with the app nrf connect I managed to update the bluetooth device

You are talking about nRF Connect for iOS, yes?

This issue seems to be related to the caching mechanism. iOS caches services of all not bonded devices if they don't have Service Changed characteristic in Generic Attribute service. It might be, that you have connected to this device address before, and the device before didn't have SC, so iOS will store the old services and after flashing a new fw onto your device the services will still be obtained from the cache. 

Could you paste here logs from the nRF Connect? And, if you have Android phone, could you also paste logs from this app when doing DFU?

philips77, 2019-01-29:
Any updates on this issue?

philips77, 2019-02-26:
Closing due to inactivity.

closed 2019-02-26 by philips77

arunjosephk, 2021-05-03:
I am getting some thing related to this issue. @philips77 could you please help me? This is the log  I got.

DFU State: DFUState
Message: [Callback] Central Manager did update state to: Powered ON
Message: Connecting to LockN12...
Message: centralManager.connect(peripheral, options: nil)
Message: [Callback] Central Manager did connect peripheral
Message: Connected to LockN12
Message: Discovering services...
Message: peripheral.discoverServices(nil)
Message: Services discovered
Message: DFU Service not found
Message: The following services were discovered:
Message:  - 1BC50001-0200-D29E-E511-446C609DB825
Message: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
Message: Disconnecting...
Message: centralManager.cancelPeripheralConnection(peripheral)
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected
ERROR: DFU Service not found

philips77, 2021-05-03:
Did you read the hint message in the log? It's all stated there. You may either:
- be connected to a wrong device -> connect to the right one, 
- the services of a firmware flashed before on this device have been cached by the iPhone -> restart Bluetooth on iPhone to clear the cache
- check if you have DFU Bootlaoder flashed on the device -> use nRF Connect (for iOS or Android) to verify that Secure DFU service is there
- if you have a custom service UUID, use [DFUUuidHelper](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUUuidHelper.swift) class to set the actual UUIDs.

arunjosephk, 2021-05-03:
@philips77  Connected to the correct device, and I have toggle the Bluetooth and  also restarted the iPhone but same message will came.

philips77, 2021-05-03:
You still have 2 points to check.

sumanmdr5, 2023-02-02:
> I am getting some thing related to this issue. @philips77 could you please help me? This is the log I got.
> 
> ```
> DFU State: DFUState
> Message: [Callback] Central Manager did update state to: Powered ON
> Message: Connecting to LockN12...
> Message: centralManager.connect(peripheral, options: nil)
> Message: [Callback] Central Manager did connect peripheral
> Message: Connected to LockN12
> Message: Discovering services...
> Message: peripheral.discoverServices(nil)
> Message: Services discovered
> Message: DFU Service not found
> Message: The following services were discovered:
> Message:  - 1BC50001-0200-D29E-E511-446C609DB825
> Message: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
> Message: Disconnecting...
> Message: centralManager.cancelPeripheralConnection(peripheral)
> Message: [Callback] Central Manager did disconnect peripheral
> Message: Disconnected
> ERROR: DFU Service not found
> ```
> 
> Here is my code
> 
> ```swift
> guard let selectedFirmware = DFUFirmware(urlToZipFile:url), let peripheral = self.peripheral else {
>    return
> }
> let initiator = DFUServiceInitiator().with(firmware: selectedFirmware)
> initiator.logger = self
> initiator.delegate = self
> initiator.progressDelegate = self
> let _ = initiator.start(target: peripheral)
> ```

am having the same issue did you find any solutions to it??

sumanmdr5, 2023-02-02:
Connecting to device...
Connected to RAT
Discovering services...
peripheral.discoverServices(nil)
Services discovered
DFU Service not found
The following services were discovered:
 - D0611E78-BBB4-4591-A5F8-487910AE4366
 - 9FA480E0-4967-4542-9390-D343DC5D04AE
 - 180A
Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
Disconnecting...
centralManager.cancelPeripheralConnection(peripheral)
[Callback] Central Manager did disconnect peripheral
Disconnected
DFU error: DFU Service not found