nordicsemi/IOS-DFU-Library #260: CRC Error

PJCSpencer, 2019-01-09:
I have another question. 

Please accept my apologies this is continuation of a thread which found itself on the wrong library:
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/162

During our challenges we've come across a number of threads on various channels which suggest some form of latency issue maybe related to BLE write/read speed. Here are a couple of examples:
https://github.com/aritchie/bluetoothle/issues/257
https://github.com/aritchie/bluetoothle/issues/257

I was wondering if this is something you have experience and if you have anything which could contribute to resolving the situation we have. Currently our iOS application we implements write with response. We have experimented with variable write 'type' without consistent success.

In attempting to cover all bases we are also investigating how our device functions.

philips77, 2019-01-09:
Before iOS 11, where the `canSendWriteWithoutResponse` method wasn't there, we had to use PRN (Packet Receipt Notifications) to synchronize sender with the received. Every n-th packed had to be ack by the target device by sending a notification. That way we were sure that the whole outgoing buffer on the phone is empty and we can continue sending. The buffer size on iOS was around 20 packets. When more packets were sent in a loop, the Bluetooth just hang and stopped responding.
The other way of doing this is using Write With Response every n-th packet. You will get `peripheralDidWriteValueFor` callback when the data were sent (you don't get this callback when writing without response).
Both methods above slows down transmission, as they use the whole connection interval, so you miss an opportunity to send many bytes, but ensure that the buffer doesn't get overflown.
Since iOS 11 there are `canSendWriteWithoutResponse` and a delegate, which we are using now, but they don't work as expected. First of all, very often the first call to `canSend...` returns *false*, for some reason and the delegate is never called after that. So we always send the first packet, no matter what this method returns. Only then it starts getting reliable, but the buffer size appears to be 1 now. Evey second call to 'canSendW...` ends with *false*. I was expecting to be able to write ~20 packets.

Other than that the speed relies on the connection interval, MTU and Data Length Extension (DLE). You need to tailor those values for your need. DLE is supported on some never iPhones, not all., or some newer iOS. I don't know.

PJCSpencer, 2019-01-09:
Just to absolutely double check, does the library send the package as per this documentation where the maximum size of each packet is 20 bytes.

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v13.0.0%2Flib_dfu_transport_serial.html

philips77, 2019-01-09:
Since SDK 15 the DFU bootloader supports now larger MTU. Prior to that 20-bytes long packets were sent, yes. The library supports all versions of DFU since SDK 4.3 (created around year 2013).

PJCSpencer, 2019-01-09:
We are using 4.2.2, does this send 20-byte packages?

philips77, 2019-01-09:
It sends longest it can. If your bootloader is vero SDK before 15 then it will only use 20-byte long packets. It takes the value from [this](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1620312-maximumwritevaluelength) method.

closed 2019-01-29 by philips77