nordicsemi/IOS-DFU-Library #261: Stuck on connect

caggiulio, 2019-01-10:
Hi! i've a problem with my code. The DFU process is stcuk on "Connecting" dtate. I've implemented the peripheral selector delegate, but not works. This is my code: 
`guard let dfuPeripheral = dfuPeripheral else {
            print("No DFU peripheral was set")
            return
        }
        
        guard let _ = selFirmware else {
            print("No firmware found. Check your Test Set")
            return
        }
        
        // Create DFU initiator with some default configuration
        
        let dfuInitiator = DFUServiceInitiator(queue: DispatchQueue(label: "Other"))
        dfuInitiator.delegate = self
        dfuInitiator.progressDelegate = self
        dfuInitiator.logger = self
        dfuInitiator.peripheralSelector = self

        if #available(iOS 11.0, macOS 10.13, *) {
            dfuInitiator.packetReceiptNotificationParameter = 0
        }
        self.dfuController = dfuInitiator.with(firmware: selFirmware!).start(target: dfuPeripheral)`

this is my selector: 
`public func select(_ peripheral: CBPeripheral, advertisementData: [String : AnyObject], RSSI: NSNumber, hint name: String?) -> Bool {
        return peripheral.identifier == dfuPeripheral.identifier
    }
    
    public func filterBy(hint dfuServiceUUID: CBUUID) -> [CBUUID]? {
        return nil
    }`

and this are the logs:
`CONNECTING
V: Connecting to SmartTag...
D: centralManager.connect(peripheral, options: nil)
State changed to: Connecting
D: [Callback] Central Manager did connect peripheral
I: Connected to SmartTag
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Legacy DFU...
I: Connected to SmartTag
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
A: Version number read: 0.1
...
State changed to: Starting
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
ENABLING DFU MODE
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
State changed to: Enabling DFU Mode
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to SmartTag...
D: centralManager.connect(peripheral, options: nil)
`

philips77, 2019-01-10:
You are comparing identifiers, which may not work. The device starts advertising with a new address after jumping to bootloader, so for iOS it's completly new device and gets a new identifier. You have to rely on data given in the advertising packet or RSSI.
Doesn't the default selector work for you? It should work if you didn't make any changes in the bootloader (didin't change the service UUID in the adv packet).

caggiulio, 2019-01-11:
indeed I noticed that with some firmware the update is successful, while others do not. Both with the default selector and with the one written by me. In all the firmware only the application is updated and never the bootloader

caggiulio, 2019-01-11:
mmm...i think that no. I have the logs of an upload successful, and seems jump to bootloader

```CONNECTING
V: Connecting to SmartTag...
D: centralManager.connect(peripheral, options: nil)
State changed to: Connecting
D: [Callback] Central Manager did connect peripheral
I: Connected to SmartTag
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Legacy DFU...
I: Connected to SmartTag
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
A: Version number read: 0.1
...
State changed to: Starting
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
ENABLING DFU MODE
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
State changed to: Enabling DFU Mode
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to SmartTag...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to SmartTag
V: Discovering services...
D: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A: Version number read: 0.8
...
State changed to: Starting
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing image sizes (0b, 0b, 131072b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x000000000000000000000200, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Start DFU (Op Code = 1, Upload Mode = 4) request sent
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
A: Response (Op Code = 1, Status = 1) received
A: Writing Initialize DFU Parameters...
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
[... 114 lines cut ...]
PROGRESS: 0.96
PROGRESS: 0.97
PROGRESS: 0.98
PROGRESS: 0.99
PROGRESS: 1.0
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100301
A: Response (Op Code = 3, Status = 1) received
A: Upload completed in 33.00 seconds
VALIDATING
State changed to: Validating
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x04, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Validate Firmware (Op Code = 4) request sent
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100401
A: Response (Op Code = 4, Status = 1) received
...
State changed to: Disconnecting
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x05, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Activate and Reset (Op Code = 5) request sent
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
COMPLETED```

caggiulio, 2019-01-11:
it happens to try 2-3 times updating and everything works

caggiulio, 2019-01-14:
any ideas?

philips77, 2019-01-15:
You may check with some other phone if your device is indeed advertising after switching when the app stuck are connecting. I would suspect that the device went into some weird state at those times.

caggiulio, 2019-01-15:
> You may check with some other phone if your device is indeed advertising after switching when the app stuck are connecting. I would suspect that the device went into some weird state at those times.
> E.g. it could start advertising without Service UUID, or with an empty packet, etc.

if so, what should I do?

philips77, 2019-01-15:
Well.. then it's a problem with the bootloader, so I would recommend creating an issue on DevZone and ask there. It will be handled by our Support Team which has much more experience with those kind of questions. I can barely compile a project in SES.
Android lib bases the reconnection on device address (bootloader get address +1), but as MAC addresses are not available through iOS API, we have to rely on this DfuPeripheralSelector.
If your DFU doesn't work on Android as well at those times, perhaps the bootloader isn't advertising at all?

caggiulio, 2019-01-17:
ok, i'll try and I'll let you know

philips77, 2019-01-29:
Hi, did you manage to solve the problem?

caggiulio, 2019-02-06:
Unfortunately not. I noticed that some firmware do not give me problems, others yes. At this point I think it's a problem of firmware

philips77, 2019-02-06:
I'll close the issue here then. Feel free to reopen it when you find it's lib related or open a new one if needed.

closed 2019-02-06 by philips77