nordicsemi/IOS-DFU-Library #262: Expose `SecureDFUExtendedErrorCode` — unable to get relevant error from client code

al-zonke, 2019-01-24:
From client code I want to distinguish different types of error. 
Especially `SecureDFUExtendedErrorCode`, that are internal and exposed just by their description value. In some cases it's reasonable to retry OTA with different parameters depending from particular error reason

philips77, 2019-01-24:
Indeed, it could be useful. Such change would break backwards compatibility, as `.extendedError` would not be thrown anymore.
Could you do necessary PR to *develop* branch? We don't have resources right now to solve this issue. Or you can retry OTA based on the description...

philips77, 2019-02-25:
Hi, I've just made a PR with requested change.
Also, keep in mind, that with #266 the library will now (starting from version 4.3.0) automatically try to send App when a remote error was received from SD+BL init packet validation (in case a ZIP file has SD(+BL)+App). Don't know if it applies to you.

philips77, 2019-02-26:
This will be released in 4.3.0.

PR #267 (2019-02-25): Extended errors exposed
This PR fixes #262.
closed 2019-02-26 by philips77