nordicsemi/IOS-DFU-Library #265: I can not pods this repository

quanyou, 2019-02-14:
hello, when I use pods to install this repository, it give me an error with 
"[!] /usr/bin/git clone https://github.com/marmelroy/Zip.git /var/folders/w1/h3bj9fxd113fl1f00qd3_wy40000gn/T/d20190214-11813-1fa7ahu --template= --single-branch --depth 1 --branch 1.1.0

Cloning into '/var/folders/w1/h3bj9fxd113fl1f00qd3_wy40000gn/T/d20190214-11813-1fa7ahu'...
error: RPC failed; curl 56 LibreSSL SSL_read: SSL_ERROR_SYSCALL, errno 54
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: unpack-objects failed"

my cocoapods version is 1.6.0.beta.2

Also, I can not download this repository

philips77, 2019-02-26:
Hi, I'm about to release a new version today. With #251 we have migrated to new zip framework, so I hope your issue will be resolved.
Sorry for the delayed response.

philips77, 2019-03-01:
4.3.0 has been released. Could you try that?

philips77, 2019-08-21:
Closing as this issue may be obsolete.

closed 2019-08-21 by philips77