nordicsemi/IOS-DFU-Library #268: cocoapods error

MartinWong-H, 2019-02-26:
hello 
I need help, and pods version is 1.6.0 not beta ,
but it will tips this error
[!] Unable to determine Swift version for the following pods:

- `Zip` does not specify a Swift version and none of the targets (`RoWrite`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.

How can i reslove it

philips77, 2019-02-26:
Hi, I'm about to publish a version today. We have merged #251 so ZipFramework will be used instead of Zip. This should cause less troubles.
Also, I updated CocoaPods to 1.6.1. Hope it will work.

philips77, 2019-02-26:
May be similar to #265.

MartinWong-H, 2019-02-26:
OK, Thanks！！！

I solved it temporarily in one way：
build settings ->  User-Defined 

and add setting with SWIFT_VERSION 4.2

closed 2019-03-01 by philips77