nordicsemi/IOS-DFU-Library #275: Small suggestion for improvement

DevAndArtist, 2019-03-03:
Could we remove the dependency for `ZIPFoundation`? By that I don't mean to completely remote it, but to split the main library into multiple parts, one `LITE` version that does not have this dependency and `FULL` that still has it. In our case we simply don't need the zip functionality and carrying around an unused ZIP framework in the app bundle is redundant for us.

philips77, 2019-03-04:
I understand your point, but there are some buts:
1. I don't think using Base64 strings is that popular that many would benefit from the split.
2. I would rather think about having DFU with a simple DFUFirmware class, without the ZipFramework dependency + another module that extends it and allows to use ZIP. I don't know if it's possible.
3. I have a lot in my backlog right now and this doesn't seem to be important task (just an improvement). So even if it's implemented, it will take long time, so PRs are welcome :)
4. That would also require us to maintain this modules in the future. I have no idea is it easy or not, never done that. Perhaps some script would solve everything (automated release, etc).

Let's keep the task here. Perhaps it will eventually be resolved.

DevAndArtist, 2019-03-04:
Well base64 is just our way to hardcode the raw FW data into the app binary, we convert it into `Data` when required. We might add `Firebase` support in the future and distribute FW updates through the web which most likely will also avoid zip files. 

It's definitely not a super important task but it would remove unnecessary dependency from our project. If I find some time in the next couple of weeks I could provide a PR, but I'm not sure how I can quickly test pods as I probably would need to create a new podspec just for testing.