nordicsemi/IOS-DFU-Library #277: Carthage built 4.3.0 does not launch

hohteri, 2019-03-05:
With 4.2.2 everything worked fine, with 4.3.0 (and after changing Zip.framework to ZIPFoundation.framework) app fails to launch due to:

dyld: Library not loaded: @rpath/ZIPFoundation.framework/ZIPFoundation
  Referenced from: /private/var/containers/Bundle/Application/FE2923AB-4657-481B-9D42-44377F251C8A/myapp.app/Frameworks/iOSDFULibrary.framework/iOSDFULibrary
  Reason: Incompatible library version: iOSDFULibrary requires version 1.0.0 or later, but ZIPFoundation provides version 0.0.0

I did remove my DerivedData folder as suggested by some stackoverflow posts but didn't help.

philips77, 2019-03-05:
Have a look at https://github.com/weichsel/ZIPFoundation/pull/91#event-2182294024

philips77, 2019-03-06:
You will also find some discussion about this error here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/273#issuecomment-468668816

closed 2019-04-23 by philips77