nordicsemi/IOS-DFU-Library #278: Feature request: Allow DFU library users to set custom bootloader name

jlubawy, 2019-03-14:
I wanted to get your thoughts about adding a feature that would allow DFU library users to specify the bootloader name that they want, rather than using the random default.

This is especially useful for cases where a device gets stuck in the bootloader (single bank failure). Currently there is no way to associate the bootloader peripheral with the application peripheral name (to make sure they're the same physical device), and iOS doesn't provide MAC addresses so that can't be used.

If there is an alternative I haven't thought about please let me know.

I created a proof-of-concept here which better demonstrates what I'm asking for: https://github.com/jlubawy/IOS-Pods-DFU-Library/commit/6e8f1c53646259074721757ef6d577a61b8c9816

philips77, 2019-03-15:
Hi, I see no reason why not to, if it's what you find useful. Your implementation looks correct. Please, submit a PR. Just add some documentation to the name in the initiator, that it's not required to set the name and a random one will be used by default.

jlubawy, 2019-03-21:
Sounds good, I'll submit a PR as soon as I have time to clean up my branch.

philips77, 2019-08-21:
Released in 4.5.0 and now in 4.5.1.

PR #304 (2019-06-14): Allow user to specify an alternative advertising name
Allows DFU library users to specify the alternative advertisement name to use, or if `nil` (default) then it generates a random name.

Closes #278.
closed 2019-08-21 by philips77