nordicsemi/IOS-DFU-Library #279: Error 8: Operation not permitted

Alims1789, 2019-03-22:
Hello.
On iphone5s (iOS 10.3.3), the following logs are displayed.
Please let me know why.
Thank you.

I: Services discovered
V: Starting Secure DFU...
I: Connected to jeegyPlay
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
dfuStateDidChange
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu63651
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753633363531, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
dfuStateDidChange
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Scanning for the DFU Bootloader...
I: DFU Bootloader found with name Jeegyplay DFU
V: Connecting to Jeegyplay DFU...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to Jeegyplay DFU
V: Discovering services...
D: peripheral.discoverServices([FE59])
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
dfuStateDidChange
V: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
A: Secure DFU Control Point notifications enabled
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60060100010000037bfbe3e7ffffff
A: Command object info (Max size = 256, Offset = 3824909059, CRC = FFFFFFE7) received
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x010188000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
A: Packet Receipt Notif disabled (Op Code = 2, Value = 0)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x1285010a3f0801123b08ff0110341a029d012000, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x280030003888cc094224080312203e047c06d2f3, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x0424311f5796607ccb77bfca33092d177bde9fde, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x96d93ce53c64480010001a404a82a487e93d974c, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x2ada0ca93f61813674c6b766401e81388836e754, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0xa64080674c81612b5f505c93e83c6f89fa809b4a, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x19403aa0b454c758f276c46335fc20c3, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
A: Command object sent (CRC = 6E4E9066)
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6003018800000066904e6e
A: Checksum (Offset = 136, CRC = 6E4E9066) received
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
A: Command object executed
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x0602, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601001000000000000000000000
A: Data object info (Max size = 4096, Offset = 0, CRC = 00000000) received
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x010200100000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
dfuStateDidChange
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
I: Data object 1/39 created
A: Uploading firmware...
V: Sending firmware to DFU Packet characteristic...
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030158020000b8ecff6d
A: Checksum (Offset = 600, CRC = 6DFFECB8) received
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600408
E: Error 8: Operation not permitted
V: Disconnecting...
D: centralManager.cancelPeripheralConnection(peripheral)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected
Error 18: Operation not permitted

philips77, 2019-03-22:
Hmm... for some reason your target DFU device received only 600 bytes of the first data object. How big your firmware is? (I could decode it from the init packet, but perhaps you know it already). 600 bytes is very small, so I suspect some issues during sending or saving the data on the peripheral. 
Could you ask your question on DevZone? Our Support Team is more qualified in such questions. 
I understand that the lib should check if it has sent the whole object before trying to execute it, so if your fw size is > 600 bytes there is an issue with the lib as well, but that would be a workaround, not fixing the root cause of the problem.

ColaColeChao, 2019-03-23:
您好，我们在iphone7和之前的手机上使用这个动态库没有问题，但是在iPhone8和iPhone8以后的手机上使用，可以更新，但是，每次更新一点就会报错Operation not permitted，之后再连接蓝牙在升级一点海辉报错Operation not permitted，不知道是不是手机读取升级包的权限问题还是其他问题，望解答。另外，在appstore上下载的官方app同样会有这个问题

philips77, 2019-08-21:
Hi, did you manage to fix your issue? Can we close it here?

philips77, 2019-10-30:
Closing the issue due to inactivity.

closed 2019-10-30 by philips77