nordicsemi/IOS-DFU-Library #281: SIGABRT when Updating from a ZIP File

KelvinEC, 2019-03-27:
A problem occurs when tries to use a Zip file to perform a firmware update. 
Provides a Valid URL to file embbed on App Bundle in order to avoid this kind of erros.

I'm currently using version 4.3.0 of IOS-PODS-DFU-LIBRARY and current Development commit of ZipFoundation.

I'm using Carthage as Dependency Manager.
[image]

gunterhager, 2019-03-28:
I just had the same issue.

KelvinEC, 2019-04-10:
Not fixed on Version 4.4.0 yet.

bmatthys, 2019-04-11:
Same issue, nothing changed in my code, only by updating Carthage I got the issue. Also using version 4.4.0 and ZipFoundation develop branch

DrAma999, 2019-04-12:
Same issue... the problem is that if I use ZipFoundation alone it works perfectly, so is something in the ZipArchive class, but it seems to be impossible to debug it. Stack traces aren't clear at all.

```
Incident Identifier: CE39A7B1-84EA-41D9-86CA-60538F63B1F4
CrashReporter Key:   00ab514ae0dcf3167b4ab29b637f30a8f498b3ba
Hardware Model:      iPad5,2
Process:             XXXX [2282]
Path:                /private/var/containers/Bundle/Application/133BD95B-F402-4E26-99D1-4A78A71F6678/XXXX.app/XXXX
Identifier:          XXXX
Version:             1555048958 (1.0)
Code Type:           ARM-64 (Native)
Role:                Non UI
Parent Process:      launchd [1]
Coalition:           XXXX [2053]

Date/Time:           2019-04-12 08:04:02.8690 +0200
Launch Time:         2019-04-12 08:04:01.7837 +0200
OS Version:          iPhone OS 12.2 (16E227)
Baseband Version:    7.55.01
Report Version:      104

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Exception Note:  EXC_CORPSE_NOTIFY
Termination Description: DYLD, Symbol not found: _$sSo13NSFileManagerC13ZIPFoundationE9unzipItem2at2to8progressy10Foundation3URLV_AJSo10NSProgressCSgtKF | Referenced from: /private/var/containers/Bundle/Application/133BD95B-F402-4E26-99D1-4A78A71F6678/XXXX.app/Frameworks/Production_BLE.framework/Production_BLE | Expected in: /private/var/containers/Bundle/Application/133BD95B-F402-4E26-99D1-4A78A71F6678/XXXX.app/Frameworks/ZIPFoundation.framework/ZIPFoundation
Triggered by Thread:  0

Thread 0 name:  Dispatch queue: com.apple.main-thread
Thread 0 Crashed:
0   dyld                          	0x0000000100d36398 __abort_with_payload + 8
1   dyld                          	0x0000000100d359d8 abort_with_payload_wrapper_internal + 100
2   dyld                          	0x0000000100d35a08 fcntl + 0
3   dyld                          	0x0000000100cf88f8 dyld::fastBindLazySymbol+ 18680 (ImageLoader**, unsigned long) + 0
4   dyld                          	0x0000000100cf8a14 dyld::fastBindLazySymbol+ 18964 (ImageLoader**, unsigned long) + 284
5   libdyld.dylib                 	0x0000000192297848 dyld_stub_binder + 60
6   Production_BLE                	0x0000000101a73fe4 0x1019a4000 + 851940
7   XXXX       	                    0x000000010079a40c 0x100764000 + 222220
8   XXXX       	                 0x000000010079a520 0x100764000 + 222496
9   UIKitCore                     	0x00000001be3e78f8 -[UIViewController loadViewIfRequired] + 1012
10  UIKitCore                     	0x00000001be3e7cfc -[UIViewController view] + 28
11  UIKitCore                     	0x00000001be9c5b3c -[UIWindow addRootViewControllerViewIfPossible] + 136
12  UIKitCore                     	0x00000001be9c60e4 -[UIWindow _setHidden:forced:] + 272
13  UIKitCore                     	0x00000001be9d69c0 -[UIWindow makeKeyAndVisible] + 48
14  UIKitCore                     	0x00000001be989838 -[UIApplication _callInitializationDelegatesForMainScene:transitionContext:] + 3532
15  UIKitCore                     	0x00000001be98ef10 -[UIApplication _runWithMainScene:transitionContext:completion:] + 1540
16  UIKitCore                     	0x00000001be253104 __111-[__UICanvasLifecycleMonitor_Compatability _scheduleFirstCommitForScene:transition:firstActivation:completion:]_block_invoke + 776
17  UIKitCore                     	0x00000001be25b69c +[_UICanvas _enqueuePostSettingUpdateTransactionBlock:] + 160
18  UIKitCore                     	0x00000001be252d88 -[__UICanvasLifecycleMonitor_Compatability _scheduleFirstCommitForScene:transition:firstActivation:completion:] + 236
19  UIKitCore                     	0x00000001be253678 -[__UICanvasLifecycleMonitor_Compatability activateEventsOnly:withContext:completion:] + 1064
20  UIKitCore                     	0x00000001be2519c4 __82-[_UIApplicationCanvas _transitionLifecycleStateWithTransitionContext:completion:]_block_invoke + 744
21  UIKitCore                     	0x00000001be25168c -[_UIApplicationCanvas _transitionLifecycleStateWithTransitionContext:completion:] + 428
22  UIKitCore                     	0x00000001be2561cc __125-[_UICanvasLifecycleSettingsDiffAction performActionsForCanvas:withUpdatedScene:settingsDiff:fromSettings:transitionContext:]_block_invoke + 220
23  UIKitCore                     	0x00000001be256fb0 _performActionsWithDelayForTransitionContext + 112
24  UIKitCore                     	0x00000001be256084 -[_UICanvasLifecycleSettingsDiffAction performActionsForCanvas:withUpdatedScene:settingsDiff:fromSettings:transitionContext:] + 244
25  UIKitCore                     	0x00000001be25ad84 -[_UICanvas scene:didUpdateWithDiff:transitionContext:completion:] + 360
26  UIKitCore                     	0x00000001be98d518 -[UIApplication workspace:didCreateScene:withTransitionContext:completion:] + 540
27  UIKitCore                     	0x00000001be589f0c -[UIApplicationSceneClientAgent scene:didInitializeWithEvent:completion:] + 360
28  FrontBoardServices            	0x00000001951bdd44 -[FBSSceneImpl _didCreateWithTransitionContext:completion:] + 440
29  FrontBoardServices            	0x00000001951c7754 __56-[FBSWorkspace client:handleCreateScene:withCompletion:]_block_invoke_2 + 256
30  FrontBoardServices            	0x00000001951c6f5c __40-[FBSWorkspace _performDelegateCallOut:]_block_invoke + 64
31  libdispatch.dylib             	0x00000001922847d4 _dispatch_client_callout + 16
32  libdispatch.dylib             	0x00000001922295d8 _dispatch_block_invoke_direct$VARIANT$mp + 224
[... cut ...]
   x12: 0x662e6e6f69746164  x13: 0x6b726f77656d6172  x14: 0x6e756f4650495a2f  x15: 0x000a6e6f69746164
   x16: 0x0000000000000209  x17: 0x0000000101acc3c0  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x000000016f6981b0  x21: 0x00000000000001b8  x22: 0x000000016f6985b0  x23: 0x0000000000000004
   x24: 0x0000000000000006  x25: 0x0000000149e0e810  x26: 0x0000000000000280  x27: 0x0000000149e0e810
   x28: 0x0000000000000001   fp: 0x000000016f698180   lr: 0x0000000100d359d8
    sp: 0x000000016f698140   pc: 0x0000000100d36398 cpsr: 0x00000000
```

philips77, 2019-04-12:
Hello, did you follow instruction how to add ZipFoundation as Carthage dependency from https://github.com/weichsel/ZIPFoundation? A specially: 

> The fetched project has to be integrated into your workspace by dragging ZIPFoundation.xcodeproj to Xcode's Project Navigator. (See official Carhage docs.)

Also, I found: https://stackoverflow.com/questions/55603453/importing-zipfoundation-in-xcode-using-carthage-not-working which may be related? It's regarding ZipFoundation and Carthage.
I have no idea whether the issue is in the DFU Library, or in ZipFoundation.
Your logs say this: Symbol not found: _$sSo13NSFileManagerC13ZIPFoundationE9unzipItem2at2to8progressy10Foundation3URLV_AJSo10NSProgressCSgtKF
which looks like ZIPFoundation.unzipItem(at:to:) is not found. As if the DFU Lib didn't see ZIPFoundation.

DrAma999, 2019-04-12:
Hello Phillips,
so the dependency situation of my project is kind of complicated.
I have a project and I have a framework that manages all BTLe stuff.
The framework has some dependencies itself installed by using Carthage, the nordic library is integrated in the framework as a project because I needed to do some customization for our device.
This framework is integrated in the main project as a project.
```
MyApp.proj
|               |-> BLEFramework.proj
|                     |
|                     |-> Nordic(just classes and stuff no .proj)
|                     |-> Carthage dependecies
|
|
|
|-> Carthage dependencies
```
Everything works fine except for the issue above, the only difference I see here is that while all my Carthage dependencies are installed using the command  `carthage update --platform iOS` instead of the one suggested in the ZipFoundation documentation that uses the `--no-build` option. As far as I know this option simply only create a checkout dir and you must integrate the project itself, while I've added the framework to dependencies both in my framework and in the main project.
I never tried but I think that in my case it won't be possible to integrate `ZipFoundation` in this way, simply because I would need to have the ZipFoundation project open both in my BTle framework and in the main application, Xcode does't allow to open the same project more that one time.
If I call any function of `ZipFoundation` in the main app they works without crashes.
So it seems that my framework is not seeing `ZipFoundation` even if I've added it as I usually add frameworks using Carthage.

DrAma999, 2019-04-12:
I will probably extend the DFUFirmware class to take a URL of an already zipped files, or that only takes a manifest and search bin and dat in the same dir

dinesharjani, 2019-04-12:
Hi @DrAma999,

Like @philips77, I also work for Nordic. I'm going to try to help as much as possible with this issue.
We internally use mostly Cocoapods, so we don't have much experience with Carthage. However, I have a question, which I was wondering if you could help us with: would it be possible for you to try to replicate the issue in a new project adding our DFU Library as the only Carthage dependency? I feel like this answer will allow us to better know if the problem is with Carthage, our library, or perhaps your project structure.

Thanks.

DrAma999, 2019-04-12:
> Hi @DrAma999,
>
> Like @philips77, I also work for Nordic. I'm going to try to help as much as possible with this issue.
> We internally use mostly Cocoapods, so we don't have much experience with [...]

Hello I did it, and the result is somehow surprising.
I think that I've done everything correctly as I did in all my project where I use Carthage.

[image]
[image]

The project build correctly but is not able to complete the copy frameworks phase.
[image]
I tried to remove each path from the list but I get the same error from `iOSDFULibrary` and `ZipFoundation`.
You can download the sample from [here](https://www.dropbox.com/s/lpzq22wvcp5muue/Test%20Nordinc.zip?dl=0).

dinesharjani, 2019-04-12:
Interesting. Thanks a lot @DrAma999,

I'll do my best to check it today.

dinesharjani, 2019-04-12:
@DrAma999 I just noticed in your screenshot that there's a space in the filepath in the folder "Test Nordic". Historically, scripts don't like spaces in paths. Can you try by renaming the folder?

DrAma999, 2019-04-12:
> @DrAma999 I just noticed in your screenshot that there's a space in the filepath in the folder "Test Nordic". Historically, scripts don't like spaces in paths. Can you try by renaming the folder?
I have different projects with spaces inside the name and never had an issue, I tried anyway renaming it and the error still remain.
[image]

philips77, 2019-04-12:
Hi, please check the issue I referenced above. Author of the ZIPFoundation library gave a good comment.
Since I got #288 moment ago, I'll release 4.4.1 with the ZIPFoundation dependency incremented to 0.9.9.

philips77, 2019-04-12:
Hi, could you try 4.4.1? Seems to be working locally... I updated the lib to use ZIPFramework 0.9.9.

DrAma999, 2019-04-12:
> Hi, could you try 4.4.1? Seems to be working locally... I updated the lib to use ZIPFramework 0.9.9.

I will try on Monday... at the moment I've solved by creating another initializer.

DrAma999, 2019-04-15:
Seems to be solved, it doesn't crash anymore

PR #290 (2019-04-12): Upgrading ZIPFoundation version to 0.9.9
This PR should fix #281. It is also fixing the Cartfile on version 0.9.9 to avoid such problems in the future.
A related issue on ZIPFoundation page is here: https://github.com/weichsel/ZIPFoundation/issues/112
closed 2019-04-19 by philips77