nordicsemi/IOS-DFU-Library #282: problem with pods install

ArkFloyd, 2019-03-31:
Hi! After trying to install pods I receive these errors and .workspace file doesn't appear. Pod file is exactly the same as in instruction. Can you help me with? 

[!] Unable to determine Swift version for the following pods:

- `ZIPFoundation` does not specify a Swift version and none of the targets (`TEST_IOS_Ble`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.
- `iOSDFULibrary` does not specify a Swift version and none of the targets (`TEST_IOS_Ble`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.

[image]

philips77, 2019-04-01:
It looks like the SWIFT_VERSION build setting must be defined in the app target prior to invoking pod install.

Check out this: https://github.com/CocoaPods/CocoaPods/issues/7770

philips77, 2019-04-12:
Hi, did you manage to solve your issue?

iwown, 2019-04-15:
@philips77  Hello, I had meet the same issue. Cocoapods advise "Add swift_version to podspec", 
 but .podspec is not a file in my project. How can I do for it ?

ArkFloyd, 2019-04-15:
@philips77, yes , your link helped me to solve it, thank you! Sorry for long silent.

ArkFloyd, 2019-04-15:
Hi @iwown ! All you need to do is : 
1) Go to your target settings -> build settings
2) Then click plus - > add user defined settings 
3) (scroll down if lost it) Name the setting as SWIFT_VERSION and set a key value as 3.2 (or other value, but this one worked for me)   
[image]
[image]

closed 2019-04-19 by philips77

iwown, 2019-04-24:
Thank you for response .I had fix this issue by update lib to *pod 'iOSDFULibrary', '~ 4.4.1* .

原始邮件

Closed #282.

yexiaofeiut, 2019-06-28:
but it's not work for me!