nordicsemi/IOS-DFU-Library #288: Crash with new data value reading

clementleroy, 2019-04-12:
Hello, since the latest release I have a crash while uploading a firmware with legacyDFU
Any help would be very appreciated as it is very blocking for a lot of our customer at the moment :) 

`Crashed: DFU
0  libswiftFoundation.dylib       0x22d13c40c specialized Data._Representation.subscript.getter + 716
1  libswiftFoundation.dylib       0x22d06b8ac Data.subdata(in:) + 72
2  iOSDFULibrary                  0x1060223f0 specialized Data.asValue<A>(offset:) (Data.swift:35)
3  iOSDFULibrary                  0x106026d70 specialized DFUControlPoint.peripheral(_:didUpdateValueFor:error:) (DFUControlPoint.swift:173)
4  iOSDFULibrary                  0x106024cc0 @objc DFUControlPoint.peripheral(_:didUpdateNotificationStateFor:error:) + 4337028288
5  CoreBluetooth                  0x204cd4b48 -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 240
6  CoreBluetooth                  0x204cd4ca8 -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 132
7  CoreBluetooth                  0x204cd078c -[CBPeripheral handleMsg:args:] + 352
8  CoreBluetooth                  0x204ccb0a0 -[CBCentralManager handleMsg:args:] + 204
9  CoreBluetooth                  0x204cd9488 __30-[CBXpcConnection _handleMsg:]_block_invoke + 60
10 libdispatch.dylib              0x1feca4a38 _dispatch_call_block_and_release + 24
11 libdispatch.dylib              0x1feca57d4 _dispatch_client_callout + 16
12 libdispatch.dylib              0x1fec4e320 _dispatch_lane_serial_drain$VARIANT$mp + 592
13 libdispatch.dylib              0x1fec4ee70 _dispatch_lane_invoke$VARIANT$mp + 480
14 libdispatch.dylib              0x1fec4e1ec _dispatch_lane_serial_drain$VARIANT$mp + 284
15 libdispatch.dylib              0x1fec4ee3c _dispatch_lane_invoke$VARIANT$mp + 428
16 libdispatch.dylib              0x1fec574a8 _dispatch_workloop_worker_thread + 596
17 libsystem_pthread.dylib        0x1fee85114 _pthread_wqthread + 304
18 libsystem_pthread.dylib        0x1fee87cd4 start_wqthread + 4`

philips77, 2019-04-12:
I'm on it. Thanks for the report. I guess there should be 1 ..< 5? But with 4.4.0 this should work, as it's using the uint32 type length.

philips77, 2019-04-12:
There should be asValue(offset: 1) instead of using subdata.

philips77, 2019-04-12:
Hi, could you try with 4.4.1? Please close the issue if it works.

clementleroy, 2019-04-12:
👍It works !
Thank you for your reactivity !

PR #289 (2019-04-12): #288 fixed
This PR fixes #288.
closed 2019-04-12 by clementleroy