nordicsemi/IOS-DFU-Library #292: Issue with notification

sawanmind, 2019-04-16:
I am getting the following error.

Error:==> Enabling notifications failed
Error:==> Error 6: The request is not supported.
Can anyone let me know what's wrong?

philips77, 2019-04-23:
I understand that your device is using non-standard UUIDs. Are you sure you are setting correct services for buttonless service and its characteristic? Seems like you are setting the DFU service and DFU characteristic UUIDs for buttonless service.
Does your characteristic have Client Characteristic Configuration Descriptor (CCCD) and supports Indications/Notifications?

sawanmind, 2019-04-24:
Hi Philips,
Thanks for replying.
I am afraid. What do you mean by non-standard UUIDs? and how shall I know characteristic have CCCD.

sawanmind, 2019-04-26:
Hi @philips77 ,
I figure out what is CCCD. Now I am getting errors like below.

application:==> Buttonless DFU notifications enabled

warning:==> Application with buttonless update found

State changed to: Enabling DFU Mode

verbose:==> Writing to characteristic E97D3B10-251D-470A-A062-FA1922DFA9A8...

debug:==> peripheral.writeValue(0x01, for: E97D3B10-251D-470A-A062-FA1922DFA9A8, type: .withResponse)

error:==> Writing to characteristic failed

error:==> Error 3: Writing is not permitted.

verbose:==> Disconnecting...

debug:==> centralManager.cancelPeripheralConnection(peripheral)

debug:==> [Callback] Central Manager did disconnect peripheral

info:==> Disconnected

sawanmind, 2019-04-26:
@philips77 
Thanks in advance.
I would like to know, what am I doing wrong?

philips77, 2019-08-21:
Hi, sorry that you had to wait that long. Did you manage to solve your problem?

philips77, 2019-10-30:
Closing the issue due to inactivity.

closed 2019-10-30 by philips77