nordicsemi/IOS-DFU-Library #295: Library compile error on Data.swift

macellari-mac, 2019-04-17:
I'm trying to test the library but I'm facing a compile error on file Data.swift

Line 36: 
$0.baseAddress!.bindMemory(to: R.self, capacity: 1).pointee
Error: /Users/luciomacellari/Projects/dfuDemo/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Utilities/Data.swift:36:13: Value of type 'UnsafePointer<_>' has no member 'baseAddress'

Line 50:
array.append(contentsOf: $0)
Error:
/Users/luciomacellari/Projects/dfuDemo/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Utilities/Data.swift:50:25: Extraneous argument label 'contentsOf:' in call

I'm compiling with Xcode 10.1

KeerthiChinivar, 2019-04-18:
For **First issue** replace 

```
return subdata(in: offset ..< offset + length).withUnsafeBytes {
            $0.baseAddress!.bindMemory(to: R.self, capacity: 1).pointee
        }
```

with 

```
 #if swift(>=5.0)
        return subdata(in: offset ..< offset + length).withUnsafeBytes { $0.load(as: T.self) }
 #else
        return subdata(in: offset ..< offset + length).withUnsafeBytes { $0.pointee }
 #endif
```

For **Second issue** solution is:

```
var hexString: String {
        var array: [UInt8] = []
        self.withUnsafeBytes {
            array.append(contentsOf: getByteArray(pointer: $0))
        }
        
        return array.reduce("") { (result, byte) -> String in
            result + String(format: "%02x", byte)
        }
    }
    
    private func getByteArray(pointer: UnsafePointer<UInt8>) -> [UInt8] {
        let buffer = UnsafeBufferPointer<UInt8>(start: pointer, count: count)
        
        return [UInt8](buffer)
    }
```

philips77, 2019-04-23:
I will fix it in 4.4.2. @KeerthiChinivar thanks for your reply!

PR #299 (2019-04-23): Adding support for older Swift version.
This PR fixes #295.
closed 2019-04-23 by philips77