nordicsemi/IOS-DFU-Library #296: Error while creating ZIP stream: 未能完成操作。（“ZIPFoundation.Archive.ArchiveError”错误 0。）

hzw0407, 2019-04-19:
It used to be okay, but suddenly it turned out to be a mistake。

What is the reason for this?

philips77, 2019-04-23:
Before 4.3.0 we were using Zip dependency for unzinping packages. We have since migrated to ZIPFoundation.
You need to provide more information. Are you using Carthage or CocoaPods?

hzw0407, 2019-04-23:
yes,I tried version 4.2 and version 4.41。Neither version works。What I use is CocoaPods。
Here is my usage code。
[image]

hzw0407, 2019-04-23:
> Before 4.3.0 we were using Zip dependency for unzinping packages. We have since migrated to ZIPFoundation.
> You need to provide more information. Are you using Carthage or CocoaPods?

yes,I tried version 4.2 and version 4.41。Neither version works。What I use is CocoaPods。
Here is my usage code。

DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:manager target:peripheral];
self.selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[NSURL fileURLWithPath:[self downloadZIP:self.firmwarePath fileName:self.newestBLEVersion]]];
initiator  =  [initiator withFirmware:self.selectedFirmware];
initiator.logger = self;
initiator.delegate = self;
initiator.progressDelegate = self;
self.serviceController = [initiator startWithTarget:peripheral];

and my firmware path looks like this
/var/mobile/Containers/Data/Application/AEFB32CF-9C1E-4F4A-9829-DA62763E026D/Documents/document/TQL-112-BT-R06.zip

philips77, 2019-04-23:
I don't know what's error 0 in ZIPFramework library. Check if your ZIP is correct. Did you use [nrf util](https://github.com/NordicSemiconductor/pc-nrfutil) to create it?

hzw0407, 2019-04-24:
> I don't know what's error 0 in ZIPFramework library. Check if your ZIP is correct. Did you use [nrf util](https://github.com/NordicSemiconductor/pc-nrfutil) to create it?

I didn't use nrf util to create it,I built it step by step。The key is that there was no problem before

philips77, 2019-04-24:
Check out the https://github.com/weichsel/ZIPFoundation project for help. I don't know why such error would appear.

hzw0407, 2019-04-24:
> Check out the https://github.com/weichsel/ZIPFoundation project for help. I don't know why such error would appear.

okay,whatever,thank you so much!

pjl1234567, 2019-05-20:
I have the same problem. Have you solved it？

pjl1234567, 2019-05-20:
Error while creating ZIP stream: 未能完成操作。（“Zip.ZipError”错误 1。）

hzw0407, 2019-05-20:
> Error while creating ZIP stream: 未能完成操作。（“Zip.ZipError”错误 1。）

My previous problem was due to the incorrect format of the file。You can check if the suffix of the following file is.Zip

pjl1234567, 2019-05-20:
/var/mobile/Containers/Data/Application/83B250F7-67AA-4A01-857E-CB1CC95C4EA8/Library/Caches/Wave-V1.3.2.zip

pjl1234567, 2019-05-20:
is  uppercase Z,Zip  ，not zip?

hzw0407, 2019-05-20:
> is uppercase Z,Zip ，not zip?

 You can change capitals to lowercase.

pjl1234567, 2019-05-20:
Mine is lowercase，Are you Chinese？

hzw0407, 2019-05-20:
> Mine is lowercase，Are you Chinese？

是的 我之前报错就是因为后缀的原因。这个多半是文件格式的原因，你可以看下用到的文件是否是正确的格式

pjl1234567, 2019-05-20:
好，谢谢。我的在iPhone6 和7上都没问题，在X上遇到这个问题。

pjl1234567, 2019-05-20:
解决啦，还真是文件的问题。从网络上下载下来的，估计是文件下载没下载好

closed 2019-08-21 by philips77

ringtailsoftware, 2021-02-10:
(for anyone else encountering this problem)
For me, it was solved by using a file:// URI instead of a cdvfile:// URI
E.g. ble.upgradeFirmware(deviceId, cordova.file.applicationDirectory + 'www/upgrade.zip', ...

Strife91, 2022-05-04:
For me the issue was that the zip files was corrupted because it was interrupted during a write. To avoid this remember to add atomic to your write operation.

`try data.write(to: folderPath.appendingPathComponent("\(fileName)"), options: [.atomic])`