nordicsemi/IOS-DFU-Library #303: Scan for Characteristics returns characteristics that don't exist.

shane-bws, 2019-05-04:
The following is a log dump from my application. 

> Got a V message: Connecting to MyDevice...
Got a D message: centralManager.connect(peripheral, options: nil)
Got a D message: [Callback] Central Manager did connect peripheral
Got a I message: Connected to  MyDevice
Got a V message: Discovering services...
Got a D message: peripheral.discoverServices(nil)
Got a I message: Services discovered
Got a V message: Starting Secure DFU...
Got a I message: Connected to  MyDevice
Got a I message: Services discovered
Got a V message: Secure DFU Service found
Got a V message: Discovering characteristics in DFU Service...
Got a D message: peripheral.discoverCharacteristics(nil, for: FE59)
Got a I message: DFU characteristics discovered
Got a I message: DFU characteristic <CBCharacteristic: 0x282fc9740, UUID = 8EC90003-F315-4F60-9FB8-838830DAEA50, properties = 0x28, value = (null), notifying = NO> 8EC90003-F315-4F60-9FB8-838830DAEA50
DFU Changed to state Starting
Got a V message: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
Got a D message: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
Got a V message: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
Got a A message: Buttonless DFU indications enabled
Got a W message: Application with buttonless update found
DFU Changed to state Enabling DFU Mode
Got a V message: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
Got a D message: peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Got a I message: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
Got a I message: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
Got a A message: Response (Op Code = 1, Status = 1) received
Got a D message: [Callback] Central Manager did disconnect peripheral
Got a I message: Disconnected by the remote device
Got a V message: Scanning for the DFU Bootloader...
Got a I message: DFU Bootloader found with name MyBootloader
Got a V message: Connecting to MyDevice...
Got a D message: centralManager.connect(peripheral, options: nil)
Got a D message: [Callback] Central Manager did connect peripheral
Got a I message: Connected to MyDevice
Got a V message: Discovering services...
Got a D message: peripheral.discoverServices([FE59])
Got a I message: Services discovered
Got a V message: Secure DFU Service found
Got a V message: Discovering characteristics in DFU Service...
Got a D message: peripheral.discoverCharacteristics(nil, for: FE59)
Got a I message: DFU characteristics discovered
**Got a I message: DFU characteristic <CBCharacteristic: 0x282fc7000, UUID = 8EC90003-F315-4F60-9FB8-838830DAEA50, properties = 0x28, value = <200101>, notifying = NO> 8EC90003-F315-4F60-9FB8-838830DAEA50**
DFU Changed to state Starting
Got a V message: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
Got a D message: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
Got a E message: Enabling indications failed
Got a E message: Error 1: The handle is invalid.
Got a V message: Disconnecting...
Got a D message: centralManager.cancelPeripheralConnection(peripheral)
Got a D message: [Callback] Central Manager did disconnect peripheral
Got a I message: Disconnected
DFU Error did occur: Enabling indications failed
> 

When I use the nRF Connect iOS application to connect to the Bootloader, I see the Secure DFU Packet (8EC90002) and Secure DFU Control Point (8EC90001) characteristics, but as can be seen in the bolded log line, my own application only discovers the 8EC90003 char. Since that characteristic doesn't exist, the DFU fails.

I know that iOS does some significant caching of Bluetooth, so I suspect this might be the issue. Couple things to note:

1.  Our Bootloader keeps the same MACAddress as when it is in normal operation, unlike the reference Nordic bootloader that increments the MAC by 1.
2. My application refers to the device by its old name ("MyDevice") even after it has switched to Bootloader, and is advertising with a different name ("MyBootloader")

Have you seen this issue before? Is there a flag I can set that will override this caching and/or set the DFU library to look for the correct characteristics explicitly?

philips77, 2019-05-04:
As you're using the Buttonless service 0x0003

> Got a I message: DFU characteristic <CBCharacteristic: 0x282fc9740, UUID = 8EC90003-F315-4F60-9FB8-838830DAEA50, properties = 0x28, value = (null), notifying = NO> 8EC90003-F315-4F60-9FB8-838830DAEA50

I assume your device does not use bonding. In that case, for the iOS not to cache your services you should make sure you have the Service Changed characteristic enabled (present) in the Generic Attribute service. Presence of this characteristic for devices without trusted relationship means that services should not be cashed, and as far as i know, on contrary to Android, iOS honors that correctly. Make sure you have this characteristic. You may use nRF Connect on Android (iOS hides this service so you'll not see it in nRF Connect on iOS).

The name in CBPeripheral is also cached and replaced with the name from Device Name characteristic after connection. Is doesn't refresh it, even is the SC char is there, as far as i know. It's better to get the device name from the adv packet during scanning.

philips77, 2019-05-04:
If you enable SC on the firmware (in both app and bootloader!), dfu should work. In some Nordic SDKs SC wasn't enabled by default.

Also, if you flash different firmwares into your DK, and one of them didn't have SC enabled and you connected it to iOS device, it will cache those services so even if another fw with have SC enabled, iOS won't know, as it's will use cached ones.

In your case, i think, the app doesn't have SC enabled.

philips77, 2019-08-21:
Hi, did you manage to solve your issue?

shane-bws, 2019-08-21:
Hi, this did solve the problem thank you.

closed 2019-08-21 by shane-bws