nordicsemi/IOS-DFU-Library #307: Bin is not reducing.

sawanmind, 2019-05-17:
Hex2Bin converter is not reducing the size of the hex file when converting to bin file due to which getting the error: Data limit exceeds.

FYI, Android Samsung micro-bit has a file named HexInputStream.java is reducing the size of the hex file while converting to the bin.
I need help to resolve this problem.

philips77, 2019-05-17:
I'll look into it. Thank you for reporting.
Can you use zip instead? Zip files, created with nrf util, contain bin files. We recommend using zip for distributing updates.

sawanmind, 2019-05-17:
Thanks you for your revert. I am getting same error with zip as its again
doesn’t compressing hex file.

philips77, 2019-05-18:
If you'll use nrf util (https://github.com/NordicSemiconductor/pc-nrfutil) (for legacy dfu use the 0.5.x branch), this tool will convert hex to bin and zip files will have bin only.

sawanmind, 2019-05-22:
Hi @philips77, I did as you said to do but still, I am getting larger than 256KB.

Have a look attached below.
[image]
[image]
[image]

the command to create pem file is

[image]

I need help to resolve this problem.
Thanks in advance @philips77

sawanmind, 2019-05-22:
Hi @philips77 <https://github.com/philips77>, I did as you said but still,
I am getting file larger than 256KB.

Have a look attached below.
[image: Screenshot 2019-05-22 at 11 30 28 AM]
<https://user-images.githubusercontent.com/22234041/58150806-60eab580-7c85-11e9-8de8-f2e1f0b7f4dd.png>
[image: Screenshot 2019-05-22 at 11 30 09 AM]
<https://user-images.githubusercontent.com/22234041/58150811-65af6980-7c85-11e9-90ee-7e480240a428.png>
[image: Screenshot 2019-05-22 at 11 29 50 AM]
<https://user-images.githubusercontent.com/22234041/58150813-67792d00-7c85-11e9-871a-1bb076992f28.png>

the command to create pem file is

[image: Screenshot 2019-05-22 at 11 34 50 AM]
<https://user-images.githubusercontent.com/22234041/58150879-a4ddba80-7c85-11e9-8cea-2016bcc19ecd.png>

Error I am getting is: Data size exceeds limit.

I need help to resolve this problem.
Thanks in advance @philips77 <https://github.com/philips77>

philips77, 2019-05-22:
Hi, could you ask this question in DevZone? It's more a question for our support team, i think.

sawanmind, 2019-05-22:
okay thanks

closed 2019-08-21 by philips77