nordicsemi/IOS-DFU-Library #308: CRC does not match!

macellari-mac, 2019-05-21:
Hi,
I'm implementing DFU on nRF52840.
I created a small test project to debug DFU and I get this CRC error at the beginning of firmware upload. DFU Lib version is 4.4.2
After some retry the firmware starts uploading and completes successfully.
Please see the attached log.
[dfu_log.txt](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/3203538/dfu_log.txt)

I tries modifying dfuInitiator.packetReceiptNotificationParameter = 12 but now I get:
Error 308: 7024 bytes were sent while 0 bytes were reported as received
Start discovery

Please help.

philips77, 2019-05-22:
Hi, what phone are you using for testing?
Try lowering the packet receipt notification to 1. iPhone X may be too fast for the flash to save data. It lower the MTU on the device.

macellari-mac, 2019-05-22:
Hi, I discovered the error with an iPhone XR and I confirm that the problem persists.
Now I'm repeating the same test with an iPhone 6S and the error does not occur.

I can try to lower packetReceiptNotificationParameter but what about the below error that occurs when I set packetReceiptNotificationParameter different than 0:
Error 308: 7024 bytes were sent while 0 bytes were reported as received

Please let me know. I'm stuck on this problem.

philips77, 2019-05-23:
It is all related to the speed how fast a phone can transfer the data. iPhone X and newer can do this faster than the DFU target can handle. This should be limited on the DFU bootloader side (smaller MTU, disabling DLE, increasing connection interval, etc.), but looks like it isn't handled properly in the SDK. As the data are sent using Write Without Response, the data may be lost before they reach the application level on the firmware, that's why there are CRC checks. If you have access to the bootloader, try lowering those settings until it starts to work.
Also, as this is a fw related issue, I recommend asking this question on DevZone, where our Support Team can handle it. They may already know the answer.
The only thing you can do in this DFU Library for iOS is to set PRNs to 1, which will slow down the upload, but at least it will work (as far as I've heard). For each packet sent there will be a notification received.

macellari-mac, 2019-05-23:
I set packetReceiptNotificationParameter to 1 and now the upload works also on iphone XR.
We will look for a firmware solution on Nordic DevZone.

ehmjaysee, 2019-07-03:
macellari-mac it seems we have same or similar issue. Did you make progress with DevZone?

closed 2019-08-21 by philips77