nordicsemi/IOS-DFU-Library #309: SPM Support

DevAndArtist, 2019-06-11:
Xcode 11 now supports Swift Packages out of the box. Our team would like to switch the package managers to a single one, therefore we need this library to support SPM.

philips77, 2019-06-11:
Hi,
We have migration to SPM in our backlog. As I'm busy with mesh right now, I hope I will do this before end of June.

DevAndArtist, 2019-06-11:
@philips77 I can also take a look at it soon-ish, but I don't know how a *potential* restructure would affect other setups like cocoapods. IIRC SPM offers an alternative path to the modules source. If I get the library to compile like that then it would be an easy thing to do.

Anything specific in the SPM related ticket from your backlog that I should know about?

philips77, 2019-06-12:
Great! 

> Anything specific in the SPM related ticket from your backlog that I should know about?

No, I didn't give any thought about that. Ideally if it worked with CocoaPods, Carthage and SPM... We could just create another version with only SPM support and deprecate others. I have to read more about SPM.

blackivory86, 2019-07-05:
Any plans on doing a new release not that this is merged?

philips77, 2019-07-08:
I was hoping to release it last week but was buried with other tasks. Nothing stopping from releasing, I'll do it after I'm back from vacations.

philips77, 2019-08-21:
Released on SPM: https://swiftpm.co/?query=NordicDFU

closed 2019-08-21 by philips77