nordicsemi/IOS-DFU-Library #312: Using the Swift language is banned in our company

guo846773272, 2019-06-23:
Hello, As the title suggests,at the present stage,Using the Swift language is banned in our company. 😂 Now we need to implement OTA functionality in Objective-C?But,I can't finish the OTA implementation, 🙃 Is there any library I can use?Thank you very much

philips77, 2019-06-23:
You may check some very, very old branches or commits, when this lib was in O-C. But most features are not supported there and there's plenty of bugs. We recommend banning objective c.

DevAndArtist, 2019-06-26:
@guo846773272 I feel really sorry for your company. You need to convince your org to switch. I think Apple will slowly fade out Objective-C usage now that Swift has ABI stability.

closed 2019-07-03 by philips77