nordicsemi/IOS-DFU-Library #314: Central Manager did disconnect peripheral

mickaelhero, 2019-06-25:
Hello, when I start the update, at a certain stage the device bluetooth disconnects, I do not understand why ..

The bluetooth device has no problem because I can update the firmware via the NRFConnect application.

log:

`
👌Connecting
🖖 [Callback] Central Manager did update state to: Powered ON
🖖 Connecting to SuperBand…
🖖 centralManager.connect(peripheral, options: nil)
🖖 [Callback] Central Manager did connect peripheral
🖖 Connected to SuperBand
🖖 Discovering services...
🖖 peripheral.discoverServices(nil)
🖖 Services discovered
🖖 Starting Secure DFU...
🖖 Connected to SuperBand
🖖 Services discovered
🖖 Secure DFU Service found
🖖 Discovering characteristics in DFU Service...
🖖 peripheral.discoverCharacteristics(nil, for: FE59)
🖖 DFU characteristics discovered
👌Starting
🖖 Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
🖖 Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Buttonless DFU indications enabled
🖖 Application with buttonless update found
👌Enabling DFU Mode
🖖 Trying setting bootloader name to Dfu13547
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x02084466753133353437, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 [Callback] Central Manager did disconnect peripheral
🖖 Disconnected by the remote device
🖖 Scanning for the DFU Bootloader...
🖖 DFU Bootloader found
🖖 Connecting to Unknown device...
🖖 centralManager.connect(peripheral, options: nil)
`

My code:
`
func startUpdateDFU(with peripheral: CBPeripheral, zipFilePath: URL, completion: DFUCompletion? = nil) {
        guard let firmware = DFUFirmware(urlToZipFile: zipFilePath) else {
            return
        }

        dfuHandler = completion

        let initiator = DFUServiceInitiator(queue: nil).with(firmware: firmware)
        initiator.forceDfu = true
        initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
        initiator.delegate = self
        initiator.progressDelegate = self
        initiator.peripheralSelector = self
        initiator.logger = self

        dfuController = initiator.start(target: peripheral)
    }
`

philips77, 2019-06-25:
Looks like your device does not understand setting advertising name. Are you using Nordic SDK 14+? Did you do any modifications to the Buttonless service?
To turn off this feature you may set 
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/6ef3aa5afe100e2087ed1cdeefad7a504b224faf/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L147-L164
to `false`. The lib will just send the Jump to Bootloader command. The downside is, that if you have 2+ devices advertising DFU service in range, the app may connect to first found, not necessary
to one that you initiated DFU with. Read more in the doc.

Also, if you are not using SDK 12.x, remove the `initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true` line.
And you may also sadly remove `initiator.forceDfu = true`, as it's being used in very, very old DFU, from SDK 7 and below.

mickaelhero, 2019-06-25:
Thank you for your quick reply!
It does not work, I changed to
`@objc public var alternativeAdvertisingNameEnabled = false`

I deleted both lines as you told me.

I do not know what's the buttonless, I did not touch anything.

Nordic sdk is installed in the bluetooth device?

mickaelhero, 2019-06-25:
**EDIT**:
It work fine set `initiator.alternativeAdvertisingNameEnabled = false`

Thanks you

closed 2019-06-25 by philips77