nordicsemi/IOS-DFU-Library #317: Scanning for the DFU Bootloader...  No response

swy680614, 2019-07-02:
When I try to upload firmware. It stop on "Scanning for the DFU Bootloader..." no any response.
And device name changed to "BLE-TEST-DFU", then I try again, select the new device name.
Then upload firmware success. Why?

swy680614, 2019-07-02:
log
Verbose: Connecting to O!vita-8a8...
Debug: centralManager.connect(peripheral, options: nil)
sensor name: LE-reserved_N
Connecting...
Debug: [Callback] Central Manager did connect peripheral
Info: Connected to O!vita-8a8
Verbose: Discovering services...
Debug: peripheral.discoverServices(nil)
Info: Services discovered
Verbose: Starting Secure DFU...
Info: Connected to O!vita-8a8
Info: Services discovered
Verbose: Secure DFU Service found
Verbose: Discovering characteristics in DFU Service...
Debug: peripheral.discoverCharacteristics(nil, for: FE59)
Info: DFU characteristics discovered
Starting DFU...
Verbose: Enabling notifications for 8E400001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.setNotifyValue(true, for: 8E400001-F315-4F60-9FB8-838830DAEA50)
Verbose: Notifications enabled for 8E400001-F315-4F60-9FB8-838830DAEA50
Application: Buttonless DFU notifications enabled
Warning: Application with buttonless update found
Verbose: Writing to characteristic 8E400001-F315-4F60-9FB8-838830DAEA50...
Enabling DFU Bootloader...
Debug: peripheral.writeValue(0x01, for: 8E400001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8E400001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8E400001-F315-4F60-9FB8-838830DAEA50, value (0x):030000
Application: Response (Op Code = 1, Status = 1) received
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected by the remote device
Verbose: Scanning for the DFU Bootloader...

philips77, 2019-07-02:
Check the DfuPeripheralSelector in DfuServiceIntiator. By default it searches for DfuTarg, or a random name when using sdk 14+. You may, however, provide your own selector and return true in other conditions. 
Also, check https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/6ef3aa5afe100e2087ed1cdeefad7a504b224faf/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L164 to disable the random name feature.

swy680614, 2019-07-03:
@philips77 your sample code do not use this param, why your sample code can upload success?After 
"Scanning for the DFU Bootloader...", why  app can find bootloader?

philips77, 2019-07-03:
Are you testing the sample app against your firmware, or the attached hex and ZIP files?

swy680614, 2019-07-04:
[https://drive.google.com/file/d/1Bcb3A4hCKbY9t7zLtS1ZZIAWXXmUPyie/view?usp=sharing](url)

log

Verbose: Connecting to O!vita-8a8...
Debug: centralManager.connect(peripheral, options: nil)
Debug: [Callback] Central Manager did connect peripheral
Info: Connected to O!vita-8a8
Verbose: Discovering services...
Debug: peripheral.discoverServices(nil)
Info: Services discovered
Verbose: Starting Secure DFU...
Info: Connected to O!vita-8a8
Info: Services discovered
Verbose: Secure DFU Service found
Verbose: Discovering characteristics in DFU Service...
Debug: peripheral.discoverCharacteristics(nil, for: FE59)
Info: DFU characteristics discovered
Verbose: Enabling notifications for 8E400001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.setNotifyValue(true, for: 8E400001-F315-4F60-9FB8-838830DAEA50)
Verbose: Notifications enabled for 8E400001-F315-4F60-9FB8-838830DAEA50
Application: Buttonless DFU notifications enabled
Warning: Application with buttonless update found
Verbose: Writing to characteristic 8E400001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x01, for: 8E400001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8E400001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8E400001-F315-4F60-9FB8-838830DAEA50, value (0x):030000
Application: Response (Op Code = 1, Status = 1) received
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected by the remote device
Verbose: Scanning for the DFU Bootloader...
Info: DFU Bootloader found with name BLE-TEST-DF
Verbose: Connecting to BLE-TEST-DFU...
Debug: centralManager.connect(peripheral, options: nil)
Debug: [Callback] Central Manager did connect peripheral
Info: Connected to BLE-TEST-DFU
Verbose: Discovering services...
Debug: peripheral.discoverServices([FE59])
Info: Services discovered
Verbose: Secure DFU Service found
Verbose: Discovering characteristics in DFU Service...
Debug: peripheral.discoverCharacteristics(nil, for: FE59)
Info: DFU characteristics discovered
Verbose: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
Verbose: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
Application: Secure DFU Control Point notifications enabled
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x0200, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
Application: Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x060000000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 030000
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x030000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 030000
Application: Packet Receipt Notif disabled (Op Code = 2, Value = 0)
Verbose: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x1284010a3e0801123a080110331a028701200028, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Verbose: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x00300038e4d501422408031220b9f9c641331c54, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Verbose: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0xcc6c69fa1c91ade027f6d2e8a954bee2ebdf386e, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Verbose: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x57ce2ab3c2480110001a4080095a265201a26fed, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Verbose: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
[... 124 lines cut ...]
Debug: peripheral.writeValue(0x01, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 030000
Application: Data object executed
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x060000000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 030000
Info: Data object 7/7 created
Application: Uploading firmware...
Verbose: Sending firmware to DFU Packet characteristic...
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x01, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
[... cut ...]
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x01, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 030000
Application: Data object executed
Application: Upload completed in 15.81 seconds
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected by the remote device
error in connection_block_invoke_2: Connection interrupted

swy680614, 2019-07-04:
@philips77 I know why. Example code used 'iOSDFULibrary', '4.1.1', my project used 'iOSDFULibrary', '4.0', I change to 4.1.1, then upload firmware success.

closed 2019-07-04 by philips77