nordicsemi/IOS-DFU-Library #318: Scanning for the DFU Bootloader...

WindSwing, 2019-07-02:
Stop at " Scanning for the DFU Bootloader...",then the device restarts and the upgrade is unsuccessful. Can you show the correct code flow? Thank you very much

log is:
文件路径:/var/mobile/Containers/Data/Application/F3E1F770-5077-42A1-A690-C09C16B0A55F/Library/Caches/nrf52832_xxaa_app_V0005_2019-6-27.zip
2019-07-02 23:35:39.316 D&P patient[3151:1606003] *****连接：服务正在连接到DFU目标
2019-07-02 23:35:39.318 D&P patient[3151:1606003] 升级日志:5: Connected to A01 band 1
2019-07-02 23:35:39.318 D&P patient[3151:1606003] 升级日志:5: Services discovered
2019-07-02 23:35:39.318 D&P patient[3151:1606003] 升级日志:1: Starting Secure DFU...
2019-07-02 23:35:39.318 D&P patient[3151:1606003] 升级日志:5: Connected to A01 band 1
2019-07-02 23:35:39.319 D&P patient[3151:1606003] 升级日志:5: Services discovered
2019-07-02 23:35:39.319 D&P patient[3151:1606003] 升级日志:1: Secure DFU Service found
2019-07-02 23:35:39.319 D&P patient[3151:1606003] 升级日志:1: Discovering characteristics in DFU Service...
2019-07-02 23:35:39.319 D&P patient[3151:1606003] 升级日志:0: peripheral.discoverCharacteristics(nil, for: FE59)
2019-07-02 23:35:39.320 D&P patient[3151:1606003] 升级日志:5: DFU characteristics discovered
2019-07-02 23:35:39.320 D&P patient[3151:1606003] *****启动：DFU服务正在初始化DFU操作
2019-07-02 23:35:39.320 D&P patient[3151:1606003] 升级日志:1: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
2019-07-02 23:35:39.321 D&P patient[3151:1606003] 升级日志:0: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
2019-07-02 23:35:39.322 D&P patient[3151:1606003] 文件保存路径:file:///var/mobile/Containers/Data/Application/F3E1F770-5077-42A1-A690-C09C16B0A55F/Library/Caches/nrf52832_xxaa_app_V0005_2019-6-27.zip,失败原因:(null)
2019-07-02 23:35:39.421 D&P patient[3151:1606003] 升级日志:1: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
2019-07-02 23:35:39.422 D&P patient[3151:1606003] 升级日志:10: Buttonless DFU indications enabled
2019-07-02 23:35:39.422 D&P patient[3151:1606003] 升级日志:15: Application with buttonless update found
2019-07-02 23:35:39.422 D&P patient[3151:1606003] *****服务正在将设备切换到DFU模式
2019-07-02 23:35:39.423 D&P patient[3151:1606003] 升级日志:1: Trying setting bootloader name to Dfu94254
2019-07-02 23:35:39.423 D&P patient[3151:1606003] 升级日志:1: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2019-07-02 23:35:39.423 D&P patient[3151:1606003] 升级日志:0: peripheral.writeValue(0x02084466753934323534, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 23:35:40.201 D&P patient[3151:1606003] 升级日志:0: [Callback] Central Manager did disconnect peripheral
2019-07-02 23:35:40.201 D&P patient[3151:1606003] 升级日志:5: Disconnected by the remote device
2019-07-02 23:35:40.201 D&P patient[3151:1606003] 升级日志:1: Scanning for the DFU Bootloader...

philips77, 2019-07-02:
Check out #317.

WindSwing, 2019-07-02:
Thank you very much ,According to your method, I have solved this problem.

closed 2019-07-03 by philips77

swy680614, 2019-07-03:
@WindSwing Can you show your sample code. I dont solved this problem yet.