nordicsemi/IOS-DFU-Library #319: Intermittent Error 8: Operation Not Permitted

ehmjaysee, 2019-07-02:
With iPhone 7 we see only one error in over 40 upgrades. With iPhone XR the first upgrade after power-up works but all subsequent upgrades fail until we power cycle the target device.  The ZIP package contains the application and settings only.  ZIP package is 44KB is size.  The error usually happens near the start but sometimes happens after several packets are transferred.  The DFU secure_bootloader comes from SDK 15.2 and we use SoftDevice 6.1.1.  Target device is 52840.  We do NOT use bonding.  See the log output below.  I plan to increate the log verbosity and do another capture soon.  Any ideas how to debug this?

2019-07-02 08:50:59.400 > DFU state: Connecting
2019-07-02 08:51:00.831 > DFU state: Starting
2019-07-02 08:51:01.012 > DFU:Buttonless DFU indications enabled
2019-07-02 08:51:01.012 > DFU:Application with buttonless update found
2019-07-02 08:51:01.013 > DFU state: Enabling DFU Mode
2019-07-02 08:51:01.072 > DFU:Response (Op Code = 2, Status = 1) received
2019-07-02 08:51:01.072 > DFU:Bootloader name changed successfully
2019-07-02 08:51:01.101 > DFU:Response (Op Code = 1, Status = 1) received
2019-07-02 08:51:04.943 > DFU state: Starting
2019-07-02 08:51:04.971 > DFU:Secure DFU Control Point notifications enabled
2019-07-02 08:51:05.002 > DFU:Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
2019-07-02 08:51:05.064 > DFU:Packet Receipt Notif disabled (Op Code = 2, Value = 0)
2019-07-02 08:51:05.065 > DFU:Command object sent (CRC = 2AF5FA37)
2019-07-02 08:51:05.092 > DFU:Checksum (Offset = 141, CRC = 2AF5FA37) received
2019-07-02 08:51:05.137 > DFU:Command object executed
2019-07-02 08:51:05.166 > DFU:Packet Receipt Notif enabled (Op Code = 2, Value = 12)
2019-07-02 08:51:05.287 > DFU:Data object info (Max size = 4096, Offset = 0, CRC = 00000000) received
2019-07-02 08:51:05.287 > DFU state: Uploading
2019-07-02 08:51:05.317 > DFU:Uploading firmware...
2019-07-02 08:51:05.513 > DFU:Checksum (Offset = 3660, CRC = 91858F4F) received
2019-07-02 08:51:05.631 > DFU:Error 8: Operation not permitted
2019-07-02 08:51:05.632 > DFU:Data object executed
2019-07-02 08:51:05.663 > DFU:Uploading firmware...
2019-07-02 08:51:05.730 > DFU error: DFUError 7024 bytes were sent while 0 bytes were reported as received

ehmjaysee, 2019-07-02:
Below is the DFU log output with full verbosity.

2019-07-02 11:41:20.137 > DFU state: Connecting
2019-07-02 11:41:20.137 > DFU:[Callback] Central Manager did update state to: Powered ON
2019-07-02 11:41:20.137 > DFU:Connecting to MI.Tracer.Angle...
2019-07-02 11:41:20.137 > DFU:centralManager.connect(peripheral, options: nil)
2019-07-02 11:41:20.924 > DFU:[Callback] Central Manager did connect peripheral
2019-07-02 11:41:20.925 > DFU:Connected to MI.Tracer.Angle
2019-07-02 11:41:20.925 > DFU:Discovering services...
2019-07-02 11:41:20.925 > DFU:peripheral.discoverServices(nil)
2019-07-02 11:41:21.282 > DFU:Services discovered
2019-07-02 11:41:21.283 > DFU:Starting Secure DFU...
2019-07-02 11:41:21.283 > DFU:Connected to MI.Tracer.Angle
2019-07-02 11:41:21.283 > DFU:Services discovered
2019-07-02 11:41:21.283 > DFU:Secure DFU Service found
2019-07-02 11:41:21.283 > DFU:Discovering characteristics in DFU Service...
2019-07-02 11:41:21.284 > DFU:peripheral.discoverCharacteristics(nil, for: FE59)
2019-07-02 11:41:21.341 > DFU:DFU characteristics discovered
2019-07-02 11:41:21.341 > DFU state: Starting
2019-07-02 11:41:21.342 > DFU:Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:21.342 > DFU:peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
2019-07-02 11:41:21.521 > DFU:Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
2019-07-02 11:41:21.522 > DFU:Buttonless DFU indications enabled
2019-07-02 11:41:21.522 > DFU:Application with buttonless update found
2019-07-02 11:41:21.522 > DFU state: Enabling DFU Mode
2019-07-02 11:41:21.522 > DFU:Trying setting bootloader name to Dfu46387
2019-07-02 11:41:21.522 > DFU:Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:21.523 > DFU:peripheral.writeValue(0x02084466753436333837, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 11:41:21.581 > DFU:Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
2019-07-02 11:41:21.582 > DFU:Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
2019-07-02 11:41:21.582 > DFU:Response (Op Code = 2, Status = 1) received
2019-07-02 11:41:21.582 > DFU:Bootloader name changed successfully
2019-07-02 11:41:21.583 > DFU:Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:21.583 > DFU:peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 11:41:21.611 > DFU:Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
2019-07-02 11:41:21.611 > DFU:Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
2019-07-02 11:41:21.612 > DFU:Response (Op Code = 1, Status = 1) received
2019-07-02 11:41:24.629 > DFU:[Callback] Central Manager did disconnect peripheral
2019-07-02 11:41:24.629 > DFU:Disconnected by the remote device
2019-07-02 11:41:24.629 > DFU:Scanning for the DFU Bootloader...
2019-07-02 11:41:24.672 > DFU:DFU Bootloader found with name Dfu46387
2019-07-02 11:41:24.673 > DFU:Connecting to Dfu39785...
2019-07-02 11:41:24.673 > DFU:centralManager.connect(peripheral, options: nil)
2019-07-02 11:41:25.414 > DFU:[Callback] Central Manager did connect peripheral
2019-07-02 11:41:25.414 > DFU:Connected to Dfu39785
2019-07-02 11:41:25.414 > DFU:Discovering services...
2019-07-02 11:41:25.415 > DFU:peripheral.discoverServices([FE59])
2019-07-02 11:41:25.473 > DFU:Services discovered
2019-07-02 11:41:25.473 > DFU:Secure DFU Service found
2019-07-02 11:41:25.474 > DFU:Discovering characteristics in DFU Service...
2019-07-02 11:41:25.474 > DFU:peripheral.discoverCharacteristics(nil, for: FE59)
2019-07-02 11:41:25.475 > DFU:DFU characteristics discovered
2019-07-02 11:41:25.475 > DFU:MTU set to 247
2019-07-02 11:41:25.475 > DFU state: Starting
2019-07-02 11:41:25.476 > DFU:Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:25.476 > DFU:peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2019-07-02 11:41:25.502 > DFU:Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2019-07-02 11:41:25.502 > DFU:Secure DFU Control Point notifications enabled
2019-07-02 11:41:25.502 > DFU:Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:25.502 > DFU:peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 11:41:25.531 > DFU:Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2019-07-02 11:41:25.532 > DFU:Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
2019-07-02 11:41:25.532 > DFU:Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
2019-07-02 11:41:25.532 > DFU:Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:25.532 > DFU:peripheral.writeValue(0x01018d000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 11:41:25.561 > DFU:Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2019-07-02 11:41:25.562 > DFU:Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2019-07-02 11:41:25.562 > DFU:Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2019-07-02 11:41:25.563 > DFU:peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 11:41:25.592 > DFU:Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
[... 33 lines cut ...]
[... cut ...]
2019-07-02 11:41:26.194 > DFU:Data object 2/11 created
2019-07-02 11:41:26.195 > DFU:Uploading firmware...
2019-07-02 11:41:26.195 > DFU:Sending firmware to DFU Packet characteristic...
2019-07-02 11:41:26.252 > DFU:Disconnecting...
2019-07-02 11:41:26.253 > DFU:centralManager.cancelPeripheralConnection(peripheral)
2019-07-02 11:41:26.253 > DFU:[Callback] Central Manager did disconnect peripheral
2019-07-02 11:41:26.254 > DFU:Disconnected
2019-07-02 11:41:26.254 > DFU error: DFUError 7024 bytes were sent while 0 bytes were reported as received

philips77, 2019-07-03:
Hi, which version of the library are you using?
Some things are very wrong here:
1. The Data object 1/11 should have 4096 bytes, and this is how many are sent. But only 3660 were received. This is the main issue, that is caused by too fast upload from the iPhone XR, that is not handled by the Dfu targ. You need to **decrease the packetReceiptNotifications in DfuServiceInitiator to a small number, like 1-10 and try which is working for you**.
2. Except from that, if the reported number of bytes is less than the object size  (3660 < 4096), the lib should try to resend the missing data if the CRC is valid, send the object again, or abort the DFU in worst case, But in your case it just continues to send 0x04 (Execute command), which should cause *Operation not permitted*, as it does.
3. After receiving this error, the process should abort, but in your case it just prints that Data object was executed, which is a lie, and continues sending object 2/11. This is very wrong.

Did you modify the library in any way?

philips77, 2019-07-03:
Also, check out #308.

ehmjaysee, 2019-07-03:
Hi, yes I agree the log seems strange. For example it seems like it tried to begin packet 2/11 before 1/11 was finished? I am using iOSDFUlibrary 4.4.1 unmodified. Here is the code:

`
            if let newImage = tracer.desiredFirmware, let selectedFirmware = DFUFirmware(urlToZipFile: newImage) 
            {
                // https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/documentation.md
                let initiator = DFUServiceInitiator(queue: nil).with(firmware: selectedFirmware)
                initiator.logger = self; // - to get log info
                initiator.delegate = self; // - to be informed about current state and errors
                initiator.progressDelegate = self; // - to show progress bar
                nordicDFU = initiator.start(targetWithIdentifier: tracer.uuid)
                if (nordicDFU == nil) {
                    commonAlert?.dismiss(animated: true, completion: { doAlert(vc: self, title: "Alert", message: "DFU failed to start")})
                }
            }
`
Issue #308 does look similar. I will try setting packetReceiptNotifications to small number and test again.

Issue is also logged on DevZone:  https://devzone.nordicsemi.com/f/nordic-q-a/49260/dfu-fails-intermittently-with-error-8-operation-not-permitted

ehmjaysee, 2019-07-03:
Just added this comment to DevZone:

@phillips77 suggested to set packetReceiptNotifications to a small number so tested on iPhoneXR with setting == 1 and now we get 6 straight upgrades passed which never happened before.  This implies the bootloader cannot handle the speed of the XR or there is some kind of flow control issues in the protocol.  However, the curious thing is the 1st upgrade after a power cycle always works and the 2nd upgrade almost always fails. This implies something is not getting set (or reset) in the bootloader when we trigger the upgrade.  The device gets into a state where a power cycle is required.

philips77, 2019-07-03:
Clearly it's an issue with the dfu bootloader. I'll close the issue here and let's follow the DevZone case.

closed 2019-07-03 by philips77