nordicsemi/IOS-DFU-Library #322: [Pod-SDK-v4.4.2]Always got an error in iOS 13 beta when start DFU

GrayLand119, 2019-07-22:


philips77, 2019-07-22:
Thanks for the report. I'll have to try it on iOS 13.

tkach, 2019-08-01:
I've got the same problem, works fine on versions prior to iOS 13, but I get this error on iOS 13 (Beta 4).
My code of starting the DFU is very similar to what @GrayLand119 posted.

Part of the logs that are printed to console when `startWithTarget:peripheral ` is called and we get the error:

```
default	12:01:10.194677 +0300	bluetoothd	Received XPC message "CBMsgIdTCCDone" from session "com.[hidden]-central-1654-4122"
default	12:01:10.194753 +0300	bluetoothd	App <private> was built with SDK c0200
default	12:01:10.194835 +0300	bluetoothd	Received XPC message "CBMsgIdRetrievePeripheralsWithIdentifiers" from session "com.[hidden]-central-1654-4122"
error	12:01:10.194902 +0300	bluetoothd	Ignoring XPC message as state is "Unknown"
default	12:01:10.195912 +0300	[hidden_app_name] Error from iOS-DFU-Library : Could not obtain peripheral instance

```

This line is *only* printed to console when app is running iOS 13 (I don't see it on older iOS versions).
`	bluetoothd	Ignoring XPC message as state is "Unknown"`

Probably something had changed in bluetoothd in iOS 13 and it's now mandatory to wait before the state `CBCentralManager` becomes `poweredOn` before calling the`retrievePeripheralsWithIdentifiers:` func [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/6ef3aa5afe100e2087ed1cdeefad7a504b224faf/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L398)?

As a workaround I've tried to set a few seconds delay between initializing the `DFUServiceInitiator`  and calling `func start(target: CBPeripheral)`, and I'm able to start the DFU without this error, however BLE communication fails later (idk maybe there are similar issues at some other parts of the library or my code) but may work for you @GrayLand119

GrayLand119, 2019-08-02:
Thanks, @tkach, It works for me.

tkach, 2019-08-02:
I'm glad it helped. 
@GrayLand119 Could you please reopen this issue as the workaround helps in testing purposes but doesn't seem to be the right solution for production use?

GrayLand119, 2019-08-02:
Ok, the issue is reopened.

philfi, 2019-08-19:
Hi just curious if there's any follow up on this? I noted this issue as well: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/313 but seem this thread has done a bit more digging into the underlying cause/possible workarounds in iOS 13 beta. Still seeing these crashes from users running iOS 13 beta 7 so seems this may be something that needs updated in the library vs a bug in iOS 13 beta. If necessary I may have some time on my plate next week to help dig into possible solutions.

philips77, 2019-08-19:
Yes, we're aware of this issue. I'll fix it this week, most probably on Wednesday.

philfi, 2019-08-19:
Thanks for the update!

philips77, 2019-08-21:
This should be now working with version 4.5.1. Thanks for the great feedback, Everyone.

closed 2019-08-21 by philips77

bojennett, 2019-10-15:
So, why can't we get 4.5.1 on CocoaPods?  It maxes out at 4.3.0.  Is CocoaPods not supported anymore?

philfi, 2019-10-15:
@bojennett It is available on Cocoapods, can you give more specific details on the issue you're seeing? You may just need to run a `repo update`, that doesn't happen automatically on `install` or `update` commands anymore.

bojennett, 2019-10-15:
Looks like I had to update cocoapods (via gem).  I had 1.7.5.  When I updated (1.8.3), it pulled the latest iOSDFULibrary, and I can confirm it is working for me now!  Thanks.

bojennett, 2019-10-15:
when I ran `pod install` with my 1.7.5 pod command, it pulled the 4.3.0.  When I updated to 1.8.3 and ran `pod install`, it pulled the 4.5.1 and swapped out Zip for ZipFoundation.

PR #327 (2019-08-21): Bugfix for iOS 13
This PR fixes #322, the issue on iOS 13 when `CBCentralManager.retrievePeripherals(withIdentifiers:)` no longer works when the manager is not in `.poweredOn` state. Retrieving was moved to inside `BaseDFUPeripheral`, when we are sure the manager is ready.

Also, this PR sets some Xcode 10.3 constants.