nordicsemi/IOS-DFU-Library #324: How to disConnect the device immediately, after OTA complete。

Xiahaiquan, 2019-08-05:
Hi guys:
After OTA complete, How to disconnect the device immediately,  When I call `- (void)cancelPeripheralConnection:(CBPeripheral *)peripheral;`, But is don't work,  I can call `- (void)scanForPeripheralsWithServices:(nullable NSArray<CBUUID *> *)serviceUUIDs options:(nullable NSDictionary<NSString *, id> *)options;` to scan the device, It  work.
 Is there some to method call in iOSDFULibrary. Any suggestions ?
Thank you!

philips77, 2019-08-05:
After the DFU is complete, the device will disconnect automatically on it's own.
You may connect using the original CBPeripheral object, had it not changed the MAC address.

Xiahaiquan, 2019-08-06:
Thank you , I will try it.

closed 2019-08-21 by philips77