nordicsemi/IOS-DFU-Library #326: Carthage built 4.5.0 ：This application’s bundle identifier does not match its code signing identifier

wongshouxin, 2019-08-16:
I used Carthage built 4.5.0, but allows  appear：This application’s bundle identifier does not match its code signing identifier！
[image]

can you help me

philips77, 2019-08-16:
Just go to the library target and change the signing identifier to own.

wongshouxin, 2019-08-16:
> Just go to the library target and change the signing identifier to own

Thanks，I have changed my team signing to my personal ID，and change new bundle Identifier，but it dont helped。could you tell me  some new way。i have try changed this  two frames once more。but the problems is always here。
[image]
[image]

philips77, 2019-08-16:
You also need to change the bundle identifier. It's above the Version: 2.6.1, and by default set to com.nordicsemi.nRFToolbox, as far as I remember. You have to set it to a name matching your certificate, like com.mycompany.AnyName.

wongshouxin, 2019-08-16:
> You also need to change the bundle identifier. It's above the Version: 2.6.1, and by default set to com.nordicsemi.nRFToolbox, as far as I remember. You have to set it to a name matching your certificate, like com.mycompany.AnyName.

I have try it again, is now  can  run well on simulators, but can't run  with  device。when run in device，same problem again。so sad，This question has been bothering me all day。

philips77, 2019-08-21:
Hello, did you manage to solve it?

wongshouxin, 2019-08-22:
> Hello, did you manage to solve it?
 i tried get ZIPFoundation folder ，built it with Xcode 10.3， and get local ZIPFoundation.framework。it now works well with my iPhone device。but also can‘t work on simulators 。I have found that the framework of ZIPFoundation is not well in Xcode10。 I'm curious why build framework with Carthage can't use。just like this ：github "weichsel/ZIPFoundation" ~> 0.9 and ”carthage update --platform iOS“ 。this is a big problems！

philips77, 2019-10-30:
There were some updates since, e.g. the ZipFoundation was bound to 0.9.9. Does it work for you now?

KelvinEC, 2019-11-12:
I tried the version 0.9.9 and the problem still there. Any way to solve this?

KelvinEC, 2019-11-14:
Any News about that?

KelvinEC, 2019-11-14:
I'll give it a try.

But to bring more information to the issue. 
The ZipFoundation integration is a little bit different. 

After that, i need to use a workspace that its not my desire.

> Carthage

> Carthage is a decentralized dependency manager.
> Installation instructions can be found in the project's README file.

> To integrate ZIPFoundation into your Xcode project using Carthage, you have to add it to your Cartfile:

>github "weichsel/ZIPFoundation" ~> 0.9
>After adding ZIPFoundation to the Cartfile, you have to fetch the sources by running:

>carthage update --no-build
>The fetched project has to be integrated into your workspace by dragging >ZIPFoundation.xcodeproj to Xcode's Project Navigator. (See official Carhage docs.)

philips77, 2019-11-15:
How about using Swift Package Manager instead?

KelvinEC, 2020-02-28:
Swift Package manager isnt an option for that. 
Why do not change this dependency or implement the code? Apparently are using small features that can be accomplished with Native iOS Dependencies.

philips77, 2020-02-29:
There was a feature request to split this library into 2 parts, one without zip, another with support for it. The native zip support is added in some recent version of iOS, that would increase the minimum version.

KelvinEC, 2020-03-26:
The idea of split into 2 parts is Good. Thanks for the replies! :)

philips77, 2020-04-04:
Reference to #275.

closed 2021-04-29 by philips77