nordicsemi/IOS-DFU-Library #330: Suggestion To Enable Your Customerrs

radarcontact, 2019-08-31:
Why isn't there a **simple** Objective-C example on how to use the DFU/OTA Updates for iOS? 

Same applies for Swift and Android. 

You guys have all these complex examples but really we just need something simple to build off of. I rather not spend hours dissecting the pieces of your code that really have no relevance to the task I'm trying to accomplish. I'm confident all your developers have to go through this same struggle which impedes them from releasing something and puts a lot of work on you dealing with the customer support and confusion. 

On the positive, you put out a great product that I'm sure entails lots of hard work. Just a suggestion to better enable your customers.

philips77, 2019-09-02:
Hi, 
The describing how to start dfu (including using O-C) is in the documentation on this GitHub repo. It can be stated with 3 lines of code, not counting dependency setup.
We have examples how to use it in the sample app here, in nRF Toolbox and Nordic Thingy app, although all of which are written in Shift. We have deprecated all our apps on O-C and migrated to Swift.

radarcontact, 2019-09-02:
Thanks for your reply. I hear that. As you mentioned, you deprecated support for Objective-C. Some of your customers haven't made the same decision.  I understand maybe you don't have too but it would be helpful if you decided to support it again. 

I also understand you have a variety of sample apps. Those apps have a lot going on. It would be nice if you had something simple. That demonstrated the core functionality with as few lines of code as possible. But not so-few that the code is cryptic. 

Just a suggestion. I know you guys/gals work hard and ultimately create a great product. Always room for improvement and this is just a gentle suggestion which may be helpful to your customers and reduce workload from people asking questions about implementations in Objective-C. Particularly with some of the new features you released like the secure boot-loader. 

All the best

closed 2019-10-30 by philips77