nordicsemi/IOS-DFU-Library #334: Can't receive error  bluetoothDisabled in  dfuError delegate method.

ldhios, 2019-09-19:
I use objective-c, DFU-Library version is 4.4.2, Never receive  bluetoothDisabled error, when I disable phone bluetooth.     dfuError delegate method cdn receive other error . why?

'''
- (void)dfuStateDidChangeTo:(enum DFUState)state{
    VSLog(@"dfuState----------- %ld",state);
    switch (state) {
        case DFUStateCompleted:
            [self callBackOtaChangeState:VSOTAState_succeed];
            break;
        default:
            break;
    }
}

- (void)dfuError:(enum DFUError)error didOccurWithMessage:(NSString *)message{
    
    VSLog(@"dfuError----------- :%@",message);
    [self callBackOtaChangeState:VSOTAState_failure];
}

- (void)dfuProgressDidChangeFor:(NSInteger)part outOf:(NSInteger)totalParts to:(NSInteger)progress currentSpeedBytesPerSecond:(double)currentSpeedBytesPerSecond avgSpeedBytesPerSecond:(double)avgSpeedBytesPerSecond{
    
    
}
- (void)logWith:(enum LogLevel)level message:(NSString *)message{
    VSLog(@"dfuLogWith----------- :%@",message);
}
'''

philips77, 2019-09-19:
What error do you get instead?

ldhios, 2019-09-20:
- (void)dfuError:(enum DFUError)error didOccurWithMessage:(NSString *)message  not call， when i disable phone blueTooth。 The IOS-Pods-DFU-Library/Example  also same issues

ldhios, 2019-09-20:
dfuError delegate is can call ，but never call when i disabled phone Bluetooth。this is my mean。

philips77, 2019-10-31:
Today I tried disabling Bluetooth and I got the error callback. When do you turn the Bluetooth off? During uploading?

philips77, 2019-11-01:
#worksforme

Please, reopen the issue, post some logs and steps to reproduce if it still doesn't work for you.

closed 2019-11-01 by philips77