nordicsemi/IOS-DFU-Library #335: iOS 13 Update failed.

ChaosYang, 2019-09-21:
The centralManager.retrievePeripherals has 0 elements. 
` guard let peripheral = self.centralManager.retrievePeripherals(withIdentifiers: [uuid]).first else {
            delegate?.dfuError(.bluetoothDisabled, didOccurWithMessage: "Could not obtain peripheral instance")
            return nil
        }`
Return nil. Update failed.

philips77, 2019-09-21:
Are you using the latest version of the library? This has been fixed recently.

ChaosYang, 2019-09-22:
> Are you using the latest version of the library? This has been fixed recently.

Yes, the version I am using is the latest ----- 4.2.2

philips77, 2019-09-22:
The latest version is 4.5 1.

ChaosYang, 2019-09-22:
> The latest version is 4.5 1.

Ohh. The latest version I searched for is 4.4.2 and I will try again with the latest version. thank you very much. But i didn't search the version 4.5.1.

philips77, 2019-09-22:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases

closed 2019-09-22 by philips77