nordicsemi/IOS-DFU-Library #337: 'iOSDFULibrary/iOSDFULibrary-Swift.h' file not found

workfel, 2019-09-25:
Hello, 

I have added your plugin with pod on ionic project. And i use a plugin(https://github.com/don/cordova-plugin-ble-central) who reference this `'iOSDFULibrary/iOSDFULibrary-Swift.h'`, but it's not found.

The pod file : 
```
# DO NOT MODIFY -- auto-generated by Apache Cordova
platform :ios, '8.0'
target 'Yuzz.it' do
	project 'Yuzz.it.xcodeproj'
	pod 'iOSDFULibrary', '~> 4.1.2'
	pod 'Intercom', '~> 5.1.0'
end
```

I do `pod install` but when i launch xcode to look the PODs on the pod `IOSDFULibrary` i have only sufix file `.swift` and `.m` not  `. h`.

What i need to do for having the `.h` files ? 

Thx

workfel, 2019-09-25:
Sorry, in fact the problem came for version dependency with ble-central plugin. 
I used a old version with a new version of xcode

closed 2019-09-25 by workfel

m4tta, 2019-11-05:
What did you do to solve this? I am running into a similar issue now with the same cordova plugin.

workfel, 2019-11-05:
Hi @m4tta , i haven't the project on my computer anymore. 
But i remember i changed the Swift version on pod file. In the `iOSDFULibrary` pod plugin.
There was a `Swift 3` and i change by 4 or 5 i don't remember.