nordicsemi/IOS-DFU-Library #339: Wrong dispatch queues used

DevAndArtist, 2019-09-27:
@philips77 just got noticed that our app was crashing due to this issue (#322). However it crashed not because the `NordicDFU` lib did error out but because the error was delivered on an unexpected dispatch queue.

One example is this code, but it's not which caused the issue, but it's a potential codepath that will cause a dispatch precondition to fail:

https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/f6896ee3460428317383c519c42db4e8329ce718/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L398

If you enter this code path but await your delegate to be delivered on the queue specified during initialization `dispatchPrecondition` will trap.

Could you please check code paths like this one and make sure they get dispatched on the right queue? It would be great if there were some unit tests that use `dispatchPrecondition(.onQueue(...))` where you check *all* possible code paths or delegate callbacks that they all get delivered on the right queue. 

Our app crashes on v4.5.0 when we get `Could not obtain peripheral instance` back from the library because the message is not delivered on the correct dispatch queue.

philips77, 2019-09-27:
Please, updated to 4.5.1. Your error will be fixed.
I'm 99% sure that the delegate is always called from right queue, except in the `DfuServiceInitiator`, but I'll double check it.
Unfortunately, we have no time to implement proper unit tests... This will have to wait until I'm done with awaiting tasks.

philips77, 2019-10-30:
Hi, I've fixed the delegates and other issue I found as well. Still no unit tests :(

philips77, 2019-11-01:
Fixed and released in 4.6.0.

closed 2019-11-01 by philips77