nordicsemi/IOS-DFU-Library #340: High DFU Failure Rate

RFz2, 2019-10-04:
Hi, 

We have recently updated to the Nordic DFU library version 4.5.1 but we're experiencing a high rate of DFU failures ever since.

When the DFU process is started, it appears to fail at sending the initial packet.

[fota_logs.txt](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/3686541/fota_logs.txt)

Any help would be appreciated.

Regards
R

philips77, 2019-10-30:
Hello,
The logs you posted here don't contain sending the init packet and show actually what follows the init packet. Do you have any other logs?

philips77, 2019-10-30:
What happens after the `Sending firmware to DFU Packet characteristic...`? Do you get any callback, or something? Does it happen on all phones/iOS versions?

RFz2, 2019-10-30:
The majority of the time it disconnects, but sometimes it hangs indefinitely. Hence no further logs after that point.
It happens on all iPhones, from 5S to XS

philips77, 2019-10-30:
This seems to be more of an issue of the bootloader or iPhones, than this library. As we have never seen such issue, it may be something with your device. Did you modify the bootloader on any way? Could you ask your question on DevZone, our Support Team may be more qualified to help you.

RFz2, 2019-11-01:
It has worked previously then suddenly a huge increase in We haven't modified the bootloader.

closed 2019-11-01 by RFz2