nordicsemi/IOS-DFU-Library #341: DFU does not work anymore since iOS 13

mickaelhero, 2019-10-10:
Since iOS 13 the dfu no longer works. I get `error 203` from DFUError 203 it's `bluetooth disabled`. Why? before ios 13 that worked.

When I try to launch the DFU from NRF Connect application. It is disconnected directly. Is there a problem with iOS 13?

This is my initiator:
```
func startUpdateDFU(with peripheral: CBPeripheral, zipFilePath: URL, completion: DFUCompletion? = nil) {
        guard let firmware = DFUFirmware(urlToZipFile: zipFilePath) else {
            return
        }

        dfuHandler = completion

        let initiator = DFUServiceInitiator(queue: nil).with(firmware: firmware)
//        initiator.forceDfu = true
        initiator.alternativeAdvertisingNameEnabled = false // Must be true, if not work set it to false!
        initiator.delegate = self
        initiator.progressDelegate = self
        initiator.peripheralSelector = self
        initiator.logger = self

        dfuController = initiator.start(target: peripheral)
    }
```

bojennett, 2019-10-14:
I, too, am having this problem.

After sending the start, I immediately get a `dfuError` with code 203 "Could not obtain peripheral instance".

The same code when running on an iOS12 (or 11, or 10) device works just fine.

bojennett, 2019-10-14:
will somebody... anybody.. from Nordic at least COMMENT on this?  This is ridiculously.

BlackMonkeyidea, 2019-10-15:
@bojennett @mickaelhero 
I found that this library has solved the problem in version 4.5.1.
I tried to update the library and can upgrade DFU normally.
#322

mickaelhero, 2019-10-15:
@BlackMonkeyidea Thank you, now it works, but it's disappointing on the part of Nordic not to have had an answer on this issue after 5 days!

bojennett, 2019-10-15:
so the problem I guess then is that cocoa pods for iOSDFULibrary seems to max out at 4.3.0

philips77, 2019-10-17:
No, just change the version on your podfile to 4.5.1.

philips77, 2019-10-17:
@mickaelhero Sorry for not replying, I'm on vacations this week. But if you'd check closed issues in this repo there already were 2 or 3 issues related to iOS 13 issue. 4.5.1 was released before 13 went public.

closed 2019-10-17 by philips77