nordicsemi/IOS-DFU-Library #342: Version 4.5.1 is still unable to adapt to swift 5.1

haiztao, 2019-10-15:
I've upgraded from Carthage to version 4.5.1, but I still can't run swift 5.1 to warn of errors.

haiztao, 2019-10-15:
> I've upgraded from Carthage to version 4.5.1, but I still can't run swift 5.1 to warn of errors.
[image]
[image]

philips77, 2019-10-17:
We'll update dfu library soon.

had23j, 2019-10-24:
Hey Philips,

Any ETA on the iOS DFU library update ?

Thanks.

philips77, 2019-10-30:
Isn't Carthage compiled on your machine? Shouldn't you just call `carthage update` command and drag new binaries from *Carthage/Build/<platform>*?
See: https://github.com/Carthage/Carthage

philips77, 2019-10-30:
Have a look at: https://stackoverflow.com/a/58621404

philips77, 2019-10-30:
@DevAndArtist, could you help here? The version 4.5.1 works for you with Carthage.

DevAndArtist, 2019-10-30:
I don't exactly know what's wrong here, you could check if you use correct Xcode in case if you have multiple versions. The error messages seems to indicate that you compile with an older version of Xcode.

https://stackoverflow.com/a/37868983

Also which Carthage version do you use? (Update it if it's outdated.)

philips77, 2019-10-30:
Thanks! @haiztao @had23j you may also check out the first comment on the link above: https://stackoverflow.com/questions/37867753/change-carthage-swift-version/37868983#comment78202856_37868983 (this is if you don't want to play with console).

haiztao, 2019-10-31:
> Thanks! @haiztao @had23j you may also check out the first comment on the link above: https://stackoverflow.com/questions/37867753/change-carthage-swift-version/37868983#comment78202856_37868983 (this is if you don't want to play with console).

haiztao, 2019-10-31:
@philips77 Hi philips77 Have you tried to update to Xcode 11 and swift 5.1.See if it fits

philips77, 2019-10-31:
Yes, I'm using Xcode 11 and Swift 5.1.

philips77, 2019-11-01:
I just released 4.6.0, which is developed in Xcode 11 with Swift 5.1.

closed 2019-11-07 by haiztao