nordicsemi/IOS-DFU-Library #343: SwiftPM import failed (Swift 5.1)

pingdan32, 2019-10-28:
import this library failed as this:
```swfit
// swift-tools-version:5.1
// The swift-tools-version declares the minimum version of Swift required to build this package.

import PackageDescription

let package = Package(
    name: "MyLibrary",
    products: [
        // Products define the executables and libraries produced by a package, and make them visible to other packages.
        .library(name: "MyLibrary", targets: ["MyLibrary"]),
    ],
    dependencies: [
        // Dependencies declare other packages that this package depends on.
        // .package(url: /* package url */, from: "1.0.0"),
        .package(url: "https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/", from: "4.5.1"),
    ],
    targets: [
        // Targets are the basic building blocks of a package. A target can define a module or a test suite.
        // Targets can depend on other targets in this package, and on products in packages which this package depends on.
        .target(name: "MyLibrary", dependencies: ["NordicDFU"]),
        .testTarget(name: "MyLibraryTests", dependencies: ["MyLibrary"]),
    ]
)

```
log: 
$ `swift package resolve`
Fetching https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/
Fetching https://github.com/weichsel/ZIPFoundation/
Completed resolution in 50.74s
Cloning https://github.com/weichsel/ZIPFoundation/
Resolving https://github.com/weichsel/ZIPFoundation/ at 0.9.9
Cloning https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/
Resolving https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/ at 4.5.1
Fatal error: 'try!' expression unexpectedly raised an error: Basic.GraphError.unexpectedCycle: file /Library/Caches/com.apple.xbs/Sources/swiftpm/swiftpm-14696/Sources/Workspace/Workspace.swift, line 1034
[2]    8353 illegal hardware instruction  swift package resolve

philips77, 2019-10-30:
I just tried to include the library to my test app (not another library) and it worked fine.

haiztao, 2019-11-11:
@pingdan32 Can swift 5.1 run after you update version 4.6.0

pingdan32, 2019-11-12:
> @pingdan32 Can swift 5.1 run after you update version 4.6.0

OK, thanks!

closed 2019-11-14 by philips77