nordicsemi/IOS-DFU-Library #345: Remote device disconnecting before upload

mzellhuber, 2019-10-30:
Peripheral is disconnecting before reaching upload. 

Here is the debug output 

```
D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to [DEVICE]...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to [DEVICE]
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to [DEVICE]
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu97462
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753937343632, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
I: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
A: Response (Op Code = 2, Status = 1) received
A: Bootloader name changed successfully
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
I: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
A: Response (Op Code = 1, Status = 1) received
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Scanning for the DFU Bootloader...
```

I've tried with iOS 13.1.3, 13.1.2, 13.2 and 12.4.1 using version 4.5.1 of the library

philips77, 2019-10-30:
Hi,
The disconnection here is intended. The device has rebooted in bootloader mode and should start advertising immediately with the name "Dfu97462". The iOS dfu library started scanning for this device using the DfuPeripheralSelector set in the initiator. By default, it will look for a device with Secure Dfu Service and the given name (numbers are randomized each time). If you've modified the bootloader to advertise with a different values, you have to have set a custom selector.
Check if your device has indeed rebooted and is advertising. I.e. check if you have flashed the bootloader on the device.

TomKeddie, 2019-10-30:
Thanks.  We're using SDK 15.3 and this works fine with Android.  I guess the rename is the big difference so I'll look into that, afaik android only uses dfutarg (probably because the mac is available to the app).

philips77, 2019-10-31:
Exactly.
On iOS you may disable this renaming feature. These should be a flag in the initiator.

philips77, 2019-11-01:
Did it help?

TomKeddie, 2019-11-01:
I didn't find the flag.  At the moment I'm focussed on why the name doesn't get from the application to the bootloader.  I'm concerned about this because it means other params probably aren't getting there.  Please close this bug, the problem is clearly in the firmware.

philips77, 2019-11-02:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d8d32fe5c4ccf229ae5d21c1055780ff259f20f0/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L187

closed 2019-11-02 by philips77