nordicsemi/IOS-DFU-Library #346: UnZip file not working.

NallapatiMalleswari, 2019-10-31:
I am downloading URL from server and saving in document directory with .zip extension. Unzip using this library not unzipping correctly with .dat, .bin, manifest.json. I am correctly saving in documentary, and I am able to unarchive and check these files from show packageContent-> AppData-> Documents -> FirmWareFiles-> latestFirmware.zip.

philips77, 2019-10-31:
Check if the files are in the root folder of the ZIP. Macs seem to put the files in a folder, not directly in the ZIP.

philips77, 2019-11-01:
Hi, did you manage to fix your issue?

NallapatiMalleswari, 2019-11-01:
Hi ,
This the code I tried to send URL to Unarchive , still I am not able to get three files after Unarchive.
func write(data: Data, to fileNamed: "Latest_Firmware", folder: String = "FirmwareFiles") {
        let fileSavingTitle = NSLocalizedString("Firmware Update", comment: "firmware file")
        let fileSavingMessage = NSLocalizedString("Saving firmware file..", comment: "firmware file")

        firmwareOTAAlert = UIAlertController(title: fileSavingTitle, message: fileSavingMessage, preferredStyle: .alert)
        DispatchQueue.main.async { [unowned self] in
            self.present(self.firmwareOTAAlert, animated: true, completion: nil)
        }
        
        guard let path = NSSearchPathForDirectoriesInDomains(.documentDirectory, .userDomainMask, true).first else { return }
        guard let writePath = NSURL(fileURLWithPath: path).appendingPathComponent(folder) else { return }
        try? FileManager.default.createDirectory(atPath: writePath.path, withIntermediateDirectories: true)
        let file = writePath.appendingPathComponent(fileNamed + ".zip")
        
        do {
            if FileManager.default.fileExists(atPath: file.absoluteString) {
                try FileManager.default.removeItem(at: file)
            }
            try data.write(to: file)
            
            guard let centralManger = BluetoothManager.shared.getCentralManager() else { return }
            guard let peripheral = BluetoothManager.shared.getPeripheral() else { return }
            self.showProgressBar()
            firmwareHelper = GloweFirmwareUpdater(with: peripheral, and: centralManger)
            firmwareHelper.delegate = self
            firmwareHelper.startUpgradeProcess(with: file)
        } catch {
            debugPrint(error.localizedDescription)
        }
    }

    func startUpgradeProcess(with fileURL: URL?) {
        guard let url = fileURL else { return }
        self.selectedFileURL  = url
       // let urlString = URL(string: "https://s3.amazonaws.com/glowe-dev-fw/Glowe_rev_2.zip")
        //let returnUrl = DFUFirmware(urlToZipFile: urlString)
        selectedFirmware = DFUFirmware(urlToZipFile: url)
        startDFUProcess()
    }
    
//    func startUpgradeProcess(with awsLink: String) {
//        guard let url = URL(string: awsLink) else { return }
//        selectedFirmware = DFUFirmware(urlToZipFile: url)
//        startDFUProcess()
//    }
    
   fileprivate func startDFUProcess() {
        guard let dfuPeripheral = self.dfuPeripheral else {
            debugPrint("No DFU peripheral was set")
            return
        }
    
        
        // This enables the experimental Buttonless DFU feature from SDK 12.
        // Please, read the field documentation before use.
        // dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
    guard let firmawre = self.selectedFirmware else {
        debugPrint("No DFU firmware was found")
        return
    }
    let dfuInitiator = DFUServiceInitiator.init().with(firmware: firmawre)
        //DFUServiceInitiator(centralManager: centralManager!, target: dfuPeripheral).with(firmware: firmawre)
    dfuInitiator.logger = self
    dfuInitiator.delegate = self
    dfuInitiator.progressDelegate = self
    dfuController = dfuInitiator.start(target: dfuPeripheral)
    
    }

philips77, 2019-11-01:
What does your `data` contain? You are creating a ZIP file by writing some data to it. Is this the firmware you want to upload, or the content of a ZIP file generated using [`nrf util`](https://github.com/NordicSemiconductor/pc-nrfutil) and obtained as `Data`?
If it's a firmware in binary, you can use this
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d8d32fe5c4ccf229ae5d21c1055780ff259f20f0/iOSDFULibrary/Classes/Implementation/Firmware/DFUFirmware.swift#L247
initiator. But remember, that for Secure DFU you need the init packet (`datFile`), so it can't be `nil`.

If you want to create a ZIP file that will work, it must contain `manifest.json` file and some other files with names described in the manifest. Have a look at the `nrf util` tool mentioned above. It can generate a valid ZIP file.

NallapatiMalleswari, 2019-11-01:
Hi,

I am getting the 'zip file' URL from AWS server. I am converting the AWS URL to data format  and then saving in document directory, and directory path I am sending to **DFUFirmware(urlToZipFile: url)** .

philips77, 2019-11-02:
This zip file, was is created with nrf util tool? You may debug why DFUFirmware returns nil.

philips77, 2019-11-02:
Also, instead of saving this Data to a file and giving URL to it, you may use this init instead: 
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d8d32fe5c4ccf229ae5d21c1055780ff259f20f0/iOSDFULibrary/Classes/Implementation/Firmware/DFUFirmware.swift#L157

NallapatiMalleswari, 2019-11-05:
Hi ,
I fixed the issue. 
Thankyou.

closed 2019-11-05 by NallapatiMalleswari