nordicsemi/IOS-DFU-Library #351: Is there an Objective-C version of the library?

ChenJian345, 2019-11-19:
I have a project using objective-c, and your library is good and I want to use it in my project. I need to use the `use_frameworks!` in my Podfile for the swift version library, but there are a lot of other libraries presented as static library. When I run `pod install`, installer will do the static library check and install failed. 
So I can **NOT** import the **iOSDFULibrary** into my project. 

Please help. Thanks.

philips77, 2020-04-04:
Hi, did you manage to solve your issue? Sorry for the long delay. I was buried with other tasks.

philips77, 2021-04-29:
I'm closing this issue due to inactivity. I think using frameworks is no longer needed for importing DFU from CocoaPods. Also, we support Carthage and SPM.

closed 2021-04-29 by philips77