nordicsemi/IOS-DFU-Library #352: swift  use ios dfu operation failed 4.6.1

lotawei, 2019-12-20:
var initiator = DFUServiceInitiator.init()
//        initiator.forceDfu = true
//        initiator.packetReceiptNotificationParameter = 12;//UInt16(UserDefaults.standard.integer(forKey: "dfu_number_of_packets"))
        initiator.logger = self
        initiator.delegate = self
        initiator.progressDelegate = self
        initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true

        initiator = initiator.with(firmware: selectedFirmware!)
        dfuController = initiator.start(target: selectedPeripheral!) 
when  enterdfu mode i always recieved Operation failed 10 
Did I use it wrong？ 
there is my log info 

name ---- Optional("DFU_E7D8")
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: [Callback] Central Manager did update state to: Powered ON
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Connecting to DFU_E7D8...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: centralManager.connect(peripheral, options: nil)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: [Callback] Central Manager did connect peripheral
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Connected to DFU_E7D8
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Discovering services...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: peripheral.discoverServices(nil)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Services discovered
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Starting Secure DFU...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Connected to DFU_E7D8
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Services discovered
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Secure DFU Service found
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Discovering characteristics in DFU Service...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: peripheral.discoverCharacteristics(nil, for: FE59)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: DFU characteristics discovered
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Application: Secure DFU Control Point notifications enabled
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6006010001000088000000a93c7d26
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Application: Command object info (Max size = 256, Offset = 136, CRC = 267D3CA9) received
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Application: Received CRC match Init packet
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60040a
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Error: Error 10: Operation failed
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
打印信息Verbose: Disconnecting...
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: centralManager.cancelPeripheralConnection(peripheral)
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Debug: [Callback] Central Manager did disconnect peripheral
/Users/lotawei/Desktop/xoss_git/XOSS_iOS/XOSS/Classes/General/Devices/SmartHardware/Wings/DFU/NORDFUViewController.swift
方法:logWith(_:message:)
行号:226
打印信息Info: Disconnected

philips77, 2019-12-20:
Which SDK is your fw based on?
If your're using SDK 12.x, the *Error 10: Operation failed* is returned when your init packet is invalid. There are several reasons: the SD ID requirement may not be fulfilled, the HW version may be different, the App version may be too low, the Bootloader version may be not greater than of the current one (assuming you're updating the bootloader), the signature may be wrong, etc. 
The bootloader from SDK 13+ returned "Extended error" in that case with a more detailed information about the cause.

philips77, 2020-04-04:
Hi, did you manage to fix your issue?

lotawei, 2020-04-04:
yes i did &nbsp;thanks

发自我的iPhone

------------------ Original ------------------
From: Aleksander Nowakowski <notifications@github.com&gt;
Date: Sat,Apr 4,2020 2:04 PM
To: NordicSemiconductor/IOS-Pods-DFU-Library <IOS-Pods-DFU-Library@noreply.github.com&gt;
Cc: lotawei <1286594907@qq.com&gt;, Author <author@noreply.github.com&gt;
Subject: Re: [NordicSemiconductor/IOS-Pods-DFU-Library] swift  use ios dfu operation failed 4.6.1 (#352)

closed 2020-04-04 by philips77