nordicsemi/IOS-DFU-Library #353: Why my iPhone must be restarted after DFU?

Hiphone-Du, 2019-12-23:
**Information**

We company make BlueTooth Toothbrush with nRF52832.there is a  question.

1.We must restart our iPhone to connect the BlueTooth Toothbrush after DFU,if we do not restart our iPhone ,we may not  connect our device, or read characteristic correctly.and any thing is doing well before DFU

2.At last,i find that, the  "FFB0" service (which is import for us) of blueTooth toothbrush  is lost after DFU,when i restart my iPhone, the  "FFB0" service appear....  

3.  Any thing is doing well by android app 

**DFU Bootloader version (please complete the following information):**
 - Bonding used: [no]
 - Library version: [4.6.1]  (by cocopod)

**Device information (please complete the following information):**
 - Device: [ iPhone 6s Plus]
 - OS: [iOS 11.2.6]

**My question**
I want not to restart my iPhone after DFU

philips77, 2019-12-23:
Are the toothbrush services changing during dfu? 
What services so you see after an update? Those that your had before the updates, or the dfu service?
Do you have Service Changed characteristic enabled on your software? This is required if you want not only to update the fw, but to change handle numbers, for example at more services or characteristics.

sky505, 2020-01-15:
I had the same problem.
But the problem is only appeared on the iphone which I used to update DFU,this issue persists before restarting the BlueTooth. However, any other  iphone can discover the characteristic normally without restarting.

philips77, 2020-01-15:
Yes, this issue is related to service cache. iOS will cache services of non bonded devices if Service Changed characteristic isn't in its attribute table. Even if you update the device, either using dfu or flashing some other app on the device, iOS will return only the cached services, as no SC means that the services may not change, so no need to do discovery again.
Dfu bootloader for non bonded devices is using a different Mac address, so from iOS perspective it's a new device, so service discovery will be performed. Depending on whether the dfu bootloader has SC, its services may also be cached. After dfu is complete, the device restarts to app mode, restoring also the normal Mac. If you see dfu services after an update, make sure your device restarts to app mode and restores the Mac address. If you ever plan to modify handles (add/remove attributes), the SC should be present. We recommend having it on both, app and bootloader mode, unless you do what you're doing. Presence of SC increases battery consumption, as service discovery is performed on every connection.

philips77, 2020-01-16:
It is not possible. Services will not be cached when Service Changed characteristic is present on a non-bonded device. If it's absent, the phone already cached services and will not clear the cache until Bluetooth is restarted in the settings. On bonded devices cache is cleared whenever a Service Changed indication is received from the device. There's no API in iOS to force-clear the cache. But you can use "tricks". For example, if your device didn't have SC, and after an update its handle numbers changed, you may start using a different Mac address (assuming address is random, not public). Then, iOS will scan a new device and will perform service discovery as for any new device.

philips77, 2020-04-04:
Can we close this issue?

closed 2020-04-07 by philips77

enseitankad0, 2024-02-09:
@philips77 
Sorry for bringing up an old topic, but I'm having the same problem.  The only difference is:
The firmware update is always successfully done, but I can't connect to the device after it finishes.
It's necessary for me to go to the bonding setting and remove my device from there. After this trick, I manage to reconnect quickly. 

The bug is not deterministic.
It's difficult to reproduce. 
It is much more likely to happen when I perform an update where the number of services and their characteristics change before and after. 
- older iPhones are more likely to experience this. 
- It never happens on an Android device.

Are there any significant changes in past sveral yers which let us to manually remove bluetooth cache from code?  If it matter we use mcumgr for flutter. Is there any way to prompt user to remove bonding manually? Thank you for your time!

philips77, 2024-02-09:
Hi,
Android does clear cache on some occasions during DFU, so after reconnection it may do a new service discovery, hence no "invalid handle" issues. On iOS the implementation is stricter, more follows the spec. When you change any of the services on the updated peripheral you need to send a Service Changed indication, which will cause an iOS device to erase cache and rediscover services. The indication shall be sent just after reconnection, or after reestablishing encryption, I'm not sure. Also, your device must have Service Changed characteristic in Generic Attribute service, otherwise iOS will cache services until the bond is removed.
Newer Bluetooth spec also added other methods of ensuring the cache is valid, e.g. Database Hash characteristic. iOS should support it by now, byt I never tried it.

enseitankad0, 2024-02-10:
Hello!

Thank you for your prompt support. I've received initial information from embedded developers that this feature is properly implemented. I want to ensure that we are correctly receiving this signal on the iOS side, and we might consider using a sniffer for this purpose.

Regarding the "Database Hash" field, this is a new element to explore. I had not come across it during troubleshooting for this issue.

I'm still wondering why this problem is not reproducible consistently. It's not that it doesn't work on the iPhone at all; rather, it seems to be a rare occurrence. Do you have any insights into whether incorrect usage of mcugr could result in issues after a reconnect?

enseitankad0, 2024-02-12:
@philips77 I've also noticed, that no matter what version of BLE is supplied on the iPhone device, error still occurs. 
Database Hash(since BLE 5.1) sounds promising, but event if we add it, there will be no effect on iPhone 13 (BLE 5.0) as this is not yet supported. 
I've tried with triggering a device rescan before reconnect (assuming that device id might have changed) but it didn't help at all.

philips77, 2024-02-14:
Your issue is related to the iPhone caching the services. I think you need to make sure the Service Change indication is sent properly and the iPhone does service discovery. It may be calling `peripheral(_:didModifyServices:)`:
https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/1518865-peripheral