nordicsemi/IOS-DFU-Library #354: I am running sample app to update firmware.I am getting device disconnection error.

NallapatiMalleswari, 2019-12-23:
Hi,
I am unable to update firmware version.While updating I am getting the below logs and it is stopped there.Could you help to fix this issue.
State changed to: Connecting
D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to GLOWE...
D: centralManager.connect(Optional("GLOWE"), options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to GLOWE
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to GLOWE
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
State changed to: Starting
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
State changed to: Enabling DFU Mode
V: Trying setting bootloader name to DfuTarg
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x020744667554617267, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to GLOWE...
D: centralManager.connect(Optional("GLOWE"), options: nil)

philips77, 2020-04-04:
For now disable the Set Advertising Name feature in Dfu Service Initiator.
I'll fix this in the next release. Have a look at #365 for more details.

philips77, 2020-04-07:
Hi,
To fix your issue, read this: https://devzone.nordicsemi.com/f/nordic-q-a/59881/advertising-rename-feature-not-working/243566#243566
Disabling Alternative Advertising Name in the `DfuServiceInitiator` can work as a workaround.

closed 2020-04-07 by philips77