nordicsemi/IOS-DFU-Library #357: Legacy DFU failing

RFz2, 2020-01-15:
Hello,

We are having trouble updating existing products running legacy DFU on NRF51 with the latest version of this library. Attached are the user logs.

[user_log.txt](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/4064542/user_log.txt)

Legacy DFU works on Android.

We are using ZIP package containing the hex and init files on iOS. The issue appears to affect the iOS Nordic Connect app also.

Any help is appreciated.

Thanks Ryan

RFz2, 2020-01-31:
Hello,

Have you experienced any issues like I explained above? It is only affecting our legacy DFU devices on iOS.

Appreciate if you could let us know if this is a known issue.

Regards
Ryan

philips77, 2020-01-31:
Nope, we didn't. We also didn't hear from anyone else... Are you able to replicate it using the samples from SDK?

philips77, 2020-04-04:
Hi, did you manage to solve your issue?

jhays, 2020-04-10:
I think I am running into this same issue. 

I am using an nRF52 dev kit board running nordic's HRM with buttonless DFU sample app on SDK 11. After I connect to it from iOS, I ask this lib to kick off the DFU process. My log looks very similar to OP's- I am stuck in an infinite loop of connect, discover, initiate, disconnect, connect.... 

If anyone has any suggestions I'd greatly appreciate it.

philips77, 2020-04-24:
What phone are you using on and what's the iOS version?
Are you using bonding?

philips77, 2020-04-24:
You device seems to disconnect on enabling notifications. It you're using bonding, and you removed bond info on one side, make sure you removed it also in the other.

jhays, 2020-04-24:
Ultimately I think my issue was misconfiguration of my dev kit board. After updating the bootloader I believe I have resolved my issue.

philips77, 2020-04-24:
Great! Good luck with the project!

closed 2020-04-24 by philips77