nordicsemi/IOS-DFU-Library #358: Guarantee word-aligned writes to the packet characteristic

SuhaibI, 2020-01-28:
After some modification of the DFU flow, I'm getting into some interesting cases where the MTU negotiation fails to guarantee word-aligned characteristic writes. Would it be possible to force this on the iOS side by changing:
```
packetSize = UInt32(characteristic.service.peripheral.maximumWriteValueLength(for: withoutResponse))
```
in SecureDFUPacket.swift line 61 to:
```
// Make the packet size the first word-aligned value that's less than the maximum
packetSize = UInt32(characteristic.service.peripheral.maximumWriteValueLength(for: .withoutResponse)) & 0xFFFC
```
I've tested this locally, and it seems to work, but I don't want to mirror and maintain this repo.

philips77, 2020-01-28:
Interesting case. Please, create the PR and we will merge it.

philips77, 2020-04-07:
What modifications did you make in the DFU?
By default, iOS sets the maximum write value length to 247, at least on iPhone 11, and this seems to be working for me. Setting `packetSize` to 244 will slow down upload speed by 1%.

philips77, 2020-04-07:
Have a look at #366. But before I merge, please clarify why it is needed and why the default behavior does not work for you.

philips77, 2020-04-24:
Ping?

SuhaibI, 2020-12-09:
Sorry - missed your pings. Commented #366. Let me know if my comment makes sense, and you have any further questions.

PR #366 (2021-01-19): Guarantee word-aligned writes
This PR fixed #358. I still need to know why it is important before merging it.
closed 2021-04-29 by philips77