nordicsemi/IOS-DFU-Library #359: Zigbee

fiveagle, 2020-01-29:
**Information**
Hi,
is it possible to use this library to update a device from a zigbee firmware to a BLE one?
I'm asking because I successfully updated a device from BLE to zigbee. But I keeping receiving the below error when I try the reverse.

Thanks.

```
E: [Callback] Central Manager failed to connect to peripheral (timeout)
Error 201: Device failed to connect
```

**DFU Bootloader version (please complete the following information):**
 - Library version: last commit

**Device information (please complete the following information):**
 - Device: iPhone XS 
 - OS:  iOS 13.3

philips77, 2020-01-29:
This is a BLE library, it relies on the device to support BLE. If updating to zigbee required also switching the soft device to one supporting zigbee, than it's not possible unless your app also supports BLE gateway and the dfu bootloader still works. Then it may be possible to trigger dfu bootloader with a button pressed during restarting. Otherwise it's not possible using this library. Perhaps you have some zigbee way of updating the device.

fiveagle, 2020-01-29:
I can press the button after restarting to enter dfu bootloader. But I see that fail to connect when le library starts the dfu update process. 
For what I know this zigbee firmware closes all connections upon receiving 2 or more connections simultaneously. So I am closing the scanning connections from CBCentralManager before starting the update, but it’s not helping.

philips77, 2020-01-30:
If I understand correctly, when you enter the bootloader with a button, the device should start advertising as BLE device with DfuTarg name and DFU service UUID (unless you modified the bootloader). The zigbee stack is in the new app and it's unused at the moment. All the zigbee features will be disabled. You should scan and connect to the bootloader and continue with DFU update. 
Also, a phone creates a single BLE connection to a device (assuming it's advertising with a single MAC address). Even if you create multiple `CBPeripheral`s that point to the same remote device, and connect all, the physical connection, seen from the remote side, will be as one. You need to disconnect all clients (`CBPeripheral`s) to disconnect the physical connection.
In your case it looks like the bootloader is not available. Make sure you scan for it, as it might have changed MAC address or no longer advertises.
Also, if your first update (BLE->zigbee) updated SoftDevice to ZigBee app, you no longer will be able to use DFU bootloader, as it relies on the SoftDevice. I'm not even sure would it work. But if you update just the app (zigbee stack is part of the app), then zigbee should work in app mode (and perhaps BLE as well if it's a mutliprotocol app, as SD is available), and BLE should work in DFU Bootloader mode (as bootloader and SoftDevice were untouched).

fiveagle, 2020-01-30:
Ok thanks,
Issue solved, it was an error the firmware side.

Thank you very much.

closed 2020-01-30 by fiveagle