nordicsemi/IOS-DFU-Library #360: Connect to other devices

mickaelhero, 2020-02-03:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [image]
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**Device information (please complete the following information):**
 - Device: iPhone 8
 - OS: 13.3

**Describe the bug**
Hello, I have a device called FATTO.

When I launch the DFU it works 1 time out of 10.

All the other times when it does not work, it is because after changing the name of the device it connects to another device (to an iMac / other device ble / unknown device, ...), or I get error 302

In the room there are some ble devices.

Why after having changed the name of the Fatto device, it disconnects and connects to another device ??

**First Log**
```
👌Connecting
🖖 [Callback] Central Manager did update state to: Powered ON
🖖 Connecting to FATTO...
🖖 centralManager.connect(peripheral, options: nil)
🖖 [Callback] Central Manager did connect peripheral
🖖 Connected to FATTO
🖖 Discovering services...
🖖 peripheral.discoverServices(nil)
🖖 Services discovered
🖖 Starting Secure DFU...
🖖 Connected to FATTO
🖖 Services discovered
🖖 Secure DFU Service found
🖖 Discovering characteristics in DFU Service...
🖖 peripheral.discoverCharacteristics(nil, for: FE59)
🖖 DFU characteristics discovered
👌Starting
🖖 Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
🖖 Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Buttonless DFU indications enabled
🖖 Application with buttonless update found
👌Enabling DFU Mode
🖖 Trying setting bootloader name to Dfu92741
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x02084466753932373431, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
🖖 Response (Op Code = 2, Status = 1) received
🖖 Bootloader name changed successfully
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
🖖 Response (Op Code = 1, Status = 1) received
imediate
🖖 [Callback] Central Manager did disconnect peripheral
🖖 Disconnected by the remote device
🖖 Scanning for the DFU Bootloader...

2020-02-03 12:31:12.207149+0200 fitto[1643:729929] [CoreBluetooth] API MISUSE: <CBPeripheral: 0x28340c3c0, identifier = 396215C7-F83E-8584-BAA3-E669AF7B95F1, name = FITTO, state = disconnected> can only accept commands while in the connected state

🖖 DFU Bootloader found
🖖 Connecting to Unknown device.. // Why does it connect to another device? 
🖖 centralManager.connect(peripheral, options: nil)
```
**Second Log**
```
👌Connecting
🖖 [Callback] Central Manager did update state to: Powered ON
🖖 Connecting to FATTO...
🖖 centralManager.connect(peripheral, options: nil)
🖖 [Callback] Central Manager did connect peripheral
🖖 Connected to FATTO
🖖 Discovering services...
🖖 peripheral.discoverServices(nil)
🖖 Services discovered
🖖 Starting Secure DFU...
🖖 Connected to FATTO
🖖 Services discovered
🖖 Secure DFU Service found
🖖 Discovering characteristics in DFU Service...
🖖 peripheral.discoverCharacteristics(nil, for: FE59)
🖖 DFU characteristics discovered
👌Starting
🖖 Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
🖖 Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Buttonless DFU indications enabled
🖖 Application with buttonless update found
👌Enabling DFU Mode
🖖 Trying setting bootloader name to MyFattoDFU // Custom name
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x02084d7920466974746f, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
🖖 Response (Op Code = 2, Status = 1) received
🖖 Bootloader name changed successfully
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
🖖 Response (Op Code = 1, Status = 1) received
🖖 [Callback] Central Manager did disconnect peripheral
🖖 Disconnected by the remote device

🖖 Scanning for the DFU Bootloader...
🖖 DFU Bootloader found with name [TV] Samsung 7 Series (43) // Why does it connect to another device?
🖖 Connecting to [TV] Samsung 7 Series (43)...
🖖 centralManager.connect(peripheral, options: nil)
```

**Third log**
```
👌Connecting
🖖 [Callback] Central Manager did update state to: Powered ON
🖖 Connecting to FATTO...
🖖 centralManager.connect(peripheral, options: nil)
🖖 [Callback] Central Manager did connect peripheral
🖖 Connected to FATTO
🖖 Discovering services...
🖖 peripheral.discoverServices(nil)
🖖 Services discovered
🖖 Starting Secure DFU...
🖖 Connected to FATTO
🖖 Services discovered
🖖 Secure DFU Service found
🖖 Discovering characteristics in DFU Service...
🖖 peripheral.discoverCharacteristics(nil, for: FE59)
🖖 DFU characteristics discovered
👌Starting
🖖 Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
🖖 Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Buttonless DFU indications enabled
🖖 Application with buttonless update found
👌Enabling DFU Mode
🖖 Trying setting bootloader name to Dfu05637
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x02084466753035363337, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
🖖 Response (Op Code = 2, Status = 1) received
🖖 Bootloader name changed successfully
🖖 Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
🖖 peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
🖖 Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
🖖 Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
🖖 Response (Op Code = 1, Status = 1) received
🖖 [Callback] Central Manager did disconnect peripheral
🖖 Disconnected by the remote device
🖖 Scanning for the DFU Bootloader...

🖖 DFU Bootloader found
🖖 Connecting to Michael 15" MacBook Pro...  // Why??
🖖 centralManager.connect(peripheral, options: nil)

2020-02-03 12:12:43.365278+0200 fitto[1629:725403] [CoreBluetooth] API MISUSE: <CBPeripheral: 0x282c2f8e0, identifier = 396215C7-F83E-8584-BAA3-E669AF7B95F1, name = FITTO, state = disconnected> can only accept commands while in the connected state

🖖 [Callback] Central Manager did connect peripheral
🖖 Connected to Michael 15" MacBook Pro // ??
🖖 Discovering services...
🖖 peripheral.discoverServices([FE59])
🖖 Services discovered
🖖 DFU Service not found
🖖 No services found
🖖 Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
🖖 Disconnecting...
🖖 centralManager.cancelPeripheralConnection(peripheral)
🖖 [Callback] Central Manager did disconnect peripheral
🖖 Disconnected
Error: 302
```

My initiator is very simple:
```
        let initiator = DFUServiceInitiator(queue: nil).with(firmware: firmware)
        initiator.delegate = self
        initiator.progressDelegate = self
        initiator.peripheralSelector = self
        initiator.logger = self
```

What is the problem?

philips77, 2020-02-03:
Hello,
Did you set your custom `DfuPeripheralSelectorDelegate`? The default one:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/b7be7a01a2a71070fb56ade7d4628ef6c6e0d162/iOSDFULibrary/Classes/Implementation/DFUPeripheralSelector.swift#L36-L53
returns *true* when names do match. Also, it should scan only for devices advertising DFU Server. In your case it looks like it tries to connect to first device found.

mickaelhero, 2020-02-03:
Got it, I implemented the protocol but with return true without logic. So if I don't implement the protocol it will run  default `DFUPeripheralSelector`?

philips77, 2020-02-03:
Yes, if you won't set the selector delegate in the initiator, the default one will be used. You should only override it if you need custom behavior. For example when your bootloader is advertising with different data, different name or you want to relay on RSSI (touch to update, etc).

closed 2020-02-03 by philips77