nordicsemi/IOS-DFU-Library #362: Fatal error:Unexpectedly found nil while unwrapping an Optional value

longhunpeter, 2020-03-02:
Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /Users/mac/iosWorkSpance/IOS-Asthma-New/asthma/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift, line 272。

philips77, 2020-03-02:
Which version of the library are you using? The latest one is 4.6.1 and there's nothing on line 272 in that file:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/b7be7a01a2a71070fb56ade7d4628ef6c6e0d162/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L270-L274

longhunpeter, 2020-03-03:
my Podfile:
`platform :ios, '8.0'
target 'asthma' do
    use_frameworks!
    pod 'iOSDFULibrary'
end`
The version obtained is 4.2.1.
change podfile:
`platform :ios, '8.0'
target 'asthma' do
    use_frameworks!
    pod 'iOSDFULibrary','4.6.1'
end`
pod install 
error:
[!] CocoaPods could not find compatible versions for pod "iOSDFULibrary":
  In Podfile:
    iOSDFULibrary (= 4.6.1)

None of your spec sources contain a spec satisfying the dependency: `iOSDFULibrary (= 4.6.1)`.

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

Note: as of CocoaPods 1.0, `pod repo update` does not happen on `pod install` by default.

philips77, 2020-03-03:
You need to update CocoaPods: https://guides.cocoapods.org/using/getting-started.html#updating-cocoapods
The new version requires `pod` in version 1.8 or something like this.
Older versions of `pod` do not support newer features and can obtain only some older version of the library.

longhunpeter, 2020-03-03:
I upgraded the pod and got 4.6.1, but using "@import iOSDFULibrary;" prompted me "Module 'iOSDFULibrary' not found".

philips77, 2020-03-03:
Did you do `pod update`?
Did you open the *xcworkspace*, not *xcodeproj* file?
Use:
```objective-c
#import <iOSDFULibrary-Swift.h>
```
Have a look at: https://medium.com/@anum.amin/swift-pod-library-in-objective-c-project-c6d1c5af997d

philips77, 2020-03-03:
If you're using Swift, use:
```swift
import iOSDFULibrary
```

longhunpeter, 2020-03-04:
I use oc.
I opened xcworkspace, using 4.2.1 is OK.
This link cannot be opened.
I look for other reasons.
thank you for your help.

philips77, 2020-03-04:
We changed the zip dependency to ZipFramework some time ago. Check on that.

longhunpeter, 2020-03-05:
I can now get 4.6.1.
Now there is another problem.
I have the following error when calling `[initiator startWithTarget: per]`:
[image]
This is my code:

        DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] withFirmware:selectedFirmware];
        initiator.delegate = self; // - to be informed about current state and errors
        initiator.progressDelegate = self; // - to show progress bar
        controller = [initiator startWithTarget:per];

longhunpeter, 2020-03-05:
Is there a demo used by oc?

longhunpeter, 2020-03-05:
I found the cause, it was a problem with my own project build settings.
Thank you for your patience.

philips77, 2020-04-04:
For you solve your issue?

closed 2020-04-07 by philips77