nordicsemi/IOS-DFU-Library #368: Error 201 connection failed

caggiulio, 2020-04-15:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [image]
Also, before asking a question, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Library version: [e.g. 4.6.1]

**Device information (please complete the following information):**
 - OS: [e.g. iOS 13.x]

**Your question**
Hi, i've a problem with version 4.6.1. Every update fails with error Error 201: Device failed to connect. This is my configuration: 

**Logs**
Error 201: Device failed to connect

caggiulio, 2020-04-16:
nothing? I tried to update with nrf connect app and i've got the same error: 201 Device failed to connect

tompi, 2020-04-16:
Im not the maintainer, but worked with the lib a lot. Can you Connect and finish the dfu process with nrf connect(nordic app that uses this library)? If yes, could you post the address that you give to the library when initiating the update?

caggiulio, 2020-04-16:
no, i can't finish the process with nrf connect :-/

tompi, 2020-04-16:
Then i would think there might be an issue with the firmware dfu implementation. Did you try discussing it with your embedded dev? Ask him if hes able to do it using nrf connect( I think he should...) again, im not really affiliated with nordic, so dont trust me blindly:)

philips77, 2020-04-23:
Hello,
Sorry for the late response. Could you post logs from the DFU process? Not just the last line, but what the lib do before.
Also, did you try the new version 4.7.1 released few days ago?

philips77, 2020-04-23:
Just a short explanation why the DFU can fail with this error:
1. Your device is not reachable, that it's not advertising. But I think iOS would not throw this error, instead it would just wait and wait until it finally is found.
2. Switching to bootloader mode does not work, and the lib is waiting for it to advertise again in bootloader mode.
3. Your device fails to connect. There may be plenty of reasons why it could fail.

Logs would help to narrow it down.

caggiulio, 2020-04-24:
I, sorry for the late response. I have to recover the logs. I will insert them in the next post. I tried with the new library but the problem is always the same

caggiulio, 2020-04-24:
`File Name: SmartTags_REL_02_15_DFU.zip
Parts: 1
Size: 131 KB
Soft Device Size: Zero KB
Bootloader Size: Zero KB
[Callback] Central Manager did update state to: Powered ON
Connecting to SmartTag...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to SmartTag
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Legacy DFU...
Connected to SmartTag
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Version number read: 0.1
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Application with buttonless update found
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
Connecting to SmartTag...
centralManager.connect(peripheral, options: nil)
centralManager.cancelPeripheralConnection(peripheral)
[Callback] Central Manager failed to connect to peripheral (timeout)`

philips77, 2020-04-24:
OK, that was helpful. 

The issue is related to Legacy DFU on non-bonded devices.
The legacy DFU, instead of using address+1 in bootloader mode, was instead using the same address and advertising directly to the phone. Have a look at these lines: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/67ada87fb5927714c6e0af7df1c9329d5792473d/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L209-L217
The direct advertising from non-bonded devices is the problem.
Recent version of Android and iOS, or recent smartphones (we didn't pinpoint it yet), use private resolvable addresses and if a peripheral performs directed advertisement, then they do not send a connection request if their MAC address has changed. It looks like either the phones change their address, so the direct advertising misses the phone, or that they stopped even responding to direct advertising from non-bonded devices. It used to work, but since few weeks we observe it on both Android and iOS world. Perhaps it was a security issue? Direct advertising from non-bonded device does not violate the spec, but can e.g. be used as DoS attack?

The possible solution is to modify the bootloader to use indirect advertising instead of direct. It's a one-liner, but to update the device you need to use an old phone, or one with older OS version. Only then new phones will be able to update the devices again.

caggiulio, 2020-04-25:
First of all, thanks for your response. Actually I've 2 problems. Randomly it show up the problem described above or another problem. Seems that my device remains in loop with this logs(from Android). 

(43ba4402-4cf5-46d8-a887-a846d9de712d, true)
D 10:29:01.071 gatt.setCharacteristicNotification(43ba4403-4cf5-46d8-a887-a846d9de712d, true)
D 10:29:01.074 gatt.setCharacteristicNotification(6e400003-b5a3-f393-e0a9-e50e24dcca9e, true)
V 10:29:44.739 Writing request to characteristic 43ba4204-4cf5-46d8-a887-a846d9de712d
D 10:29:44.739 gatt.writeCharacteristic(43ba4204-4cf5-46d8-a887-a846d9de712d, value=0xD204)
I 10:29:45.443 Data written to 43ba4204-4cf5-46d8-a887-a846d9de712d, value: (0x) D2-04
A 10:29:45.443 "(0x) D2-04" sent
V 10:29:55.912 [DFU] DFU service started
V 10:29:55.912 [DFU] Opening file...
I 10:29:55.929 [DFU] Firmware file opened successfully
V 10:29:55.929 [DFU] Connecting to DFU target...
D 10:29:55.945 [DFU] gatt = device.connectGatt(autoConnect = false)
I 10:29:55.963 [DFU] Connected to C7:01:C2:59:D4:CB
V 10:29:55.980 [DFU] Discovering services...
D 10:29:55.980 [DFU] gatt.discoverServices()
I 10:29:55.980 [DFU] Services discovered
D 10:29:56.002 [DFU] wait(1000)
V 10:29:57.012 [DFU] Reading DFU version number...
D 10:29:57.012 [DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I 10:29:57.370 [DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 01-00
A 10:29:57.370 [DFU] Version number read: 0.1
W 10:29:57.370 [DFU] Application with buttonless update found
V 10:29:57.370 [DFU] Jumping to the DFU Bootloader...
V 10:29:57.370 [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D 10:29:57.370 [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D 10:29:57.370 [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I 10:29:58.067 [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V 10:29:58.068 [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A 10:29:58.068 [DFU] Notifications enabled
D 10:29:58.068 [DFU] wait(1000)
V 10:29:59.075 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 10:29:59.075 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
D 10:30:03.119 [Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I 10:30:03.120 Disconnected
A 10:30:03.149 [DFU] Jump to bootloader sent (Op Code = 1, Upload Mode = 4)
I 10:30:03.151 [DFU] Disconnected by the remote device
D 10:30:03.151 [DFU] gatt.refresh() (hidden)
D 10:30:03.151 [DFU] gatt.close()
D 10:30:03.172 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D 10:30:03.182 [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
V 10:30:03.215 [DFU] DFU service started
I 10:30:03.215 [DFU] Firmware file opened successfully
D 10:30:03.215 [DFU] wait(1000)
D 10:30:04.215 [DFU] wait(1000)
V 10:30:05.205 [DFU] Connecting to DFU target...
D 10:30:05.260 [DFU] gatt = device.connectGatt(autoConnect = false)
I 10:30:05.492 [DFU] Connected to C7:01:C2:59:D4:CB
V 10:30:05.492 [DFU] Discovering services...
D 10:30:05.492 [DFU] gatt.discoverServices()
D 10:30:05.492 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D 10:30:05.508 [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I 10:30:07.035 [DFU] Services discovered
D 10:30:07.036 [DFU] wait(1000)
V 10:30:08.016 [DFU] Reading DFU version number...
D 10:30:08.016 [DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I 10:30:08.424 [DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 01-00
A 10:30:08.424 [DFU] Version number read: 0.1
W 10:30:08.424 [DFU] Application with buttonless update found
V 10:30:08.424 [DFU] Jumping to the DFU Bootloader...
V 10:30:08.424 [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D 10:30:08.424 [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D 10:30:08.424 [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I 10:30:09.121 [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V 10:30

:09.121 [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A 10:30:09.121 [DFU] Notifications enabled
D 10:30:09.121 [DFU] wait(1000)
[... 84 lines cut ...]
D 10:33:14.627 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I 10:33:14.627 [DFU] Connected to C7:01:C2:59:D4:CB
V 10:33:14.627 [DFU] Discovering services...
D 10:33:14.644 [DFU] gatt.discoverServices()
D 10:33:14.660 [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I 10:33:16.119 [DFU] Services discovered
D 10:33:16.121 [DFU] wait(1000)
V 10:33:17.136 [DFU] Reading DFU version number...
D 10:33:17.136 [DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I 10:33:17.493 [DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 01-00
A 10:33:17.493 [DFU] Version number read: 0.1
W 10:33:17.493 [DFU] Application with buttonless update found
V 10:33:17.493 [DFU] Jumping to the DFU Bootloader...
V 10:33:17.493 [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D 10:33:17.493 [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
[... cut ...]
I 10:33:18.200 [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V 10:33:18.200 [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A 10:33:18.200 [DFU] Notifications enabled
D 10:33:18.200 [DFU] wait(1000)
V 10:33:19.207 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 10:33:19.207 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
A 10:33:19.594 [DFU] Jump to bootloader sent (Op Code = 1, Upload Mode = 4)

The 2 problems are connected?

philips77, 2020-04-27:
No. The issue I described would happen 100% of time and could not be workaround. In the above log it looks like your device fails to switch to the bootloader mode. This may happen e.g. when you forget to flash it, and there's only SD+App (with buttonless service) on the device. After a jump the app launches again, over and over.

espiriki, 2020-05-11:
Has anyone seen this problem with older versions of Android as well? We have a Samsung Galaxy with Android 9 and an Asus Zenfone 3 with Android 8, and we are having the same problems, the DFU operation doesn't even start with the NRF Go app after we selected the .zip file

RMiron-Ember, 2020-05-22:
Hi guys,
After running into this issue, I determined that the problem is the 10 second 'connectionTimer' in the 'connect()' method in the DFUPeripheral.swift file. After increasing the timer to 25 seconds, I no longer get the 201 error. This line in the logs '[Callback] Central Manager failed to connect to peripheral (timeout)' points you in the right direction. I hope this helps.

philips77, 2020-06-16:
@RMiron-Ember are you sure you were having the same issue?
1. Legacy DFU
2. iOS 13.x
3. The buttonless service is used to jump to bootloader mode, but the library can't connect to the bootloader and times out.
4. Bootloader advertises directly using the same MAC address

From the bootloader code we see that by default it advertises directly for 50 high duty cycles, 1.28 sec each, after that time it reboots. So increasing the connection timeout should not help.

Shy0909, 2020-07-21:
File Name: RB02-040-2-20200717-SWV2.0.zip
Parts: 1
Size: 80 KB
Soft Device Size: Zero KB
Bootloader Size: Zero KB
[Callback] Central Manager did update state to: Powered ON
Connecting to RB02040C3E55C5F86D9...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to RB02040C3E55C5F86D9
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Legacy DFU...
Connected to RB02040C3E55C5F86D9
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Version number read: 0.1
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Application with buttonless update found
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
Connecting to RB02040C3E55C5F86D9...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to RB02040C3E55C5F86D9
Discovering services...
peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Reading DFU Version characteristic failed
Error 1: The handle is invalid.
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Reset (Op Code = 6) request sent
Device disconnected before sending ACK
Error 3: Writing is not permitted.
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
hi, what casue my error?

ishashkov16, 2020-10-09:
Hi @philips77 ,

I am having a random issue when updating my Nordic device.
For some reason, the device is disconnected, and then after timeout throws 201 failed to connect error:

*** Firmware: D: [Callback] Central Manager did disconnect peripheral
*** Firmware: I: Disconnected by the remote device
*** DFU State Did Change: Connecting
*** dfuStateDidChange Connecting
*** Firmware: V: Connecting to AWE...
*** Firmware: D: centralManager.connect(peripheral, options: nil)
*** Firmware: W: Connection timeout!
*** Firmware: D: centralManager.cancelPeripheralConnection(peripheral)

What is the remote device and why it can interrupt the firmware update process?
I'm using the latest SDK.
See the attached log file.
[201_failed_to_connect.pdf](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/5356181/201_failed_to_connect.pdf)

philips77, 2020-10-09:
Remote device is the Dfu target. Don't know why it resets and why doesn't start advertising again.

espoal, 2020-10-25:
I'm experiencing a similar problem, only on old devices with iOS.
Anyone found a solution?

loves-bruins, 2021-03-29:
I too am getting the "fail to load bootloader" with 4.9.0 and nRF Connect application for iOS.  Log is here from nRF connect:

> File Name: OreoPumpRadio_D9Engineering.zip
Parts: 2
Size: 222 KB
Soft Device Size: 201 KB
Bootloader Size: 1 byte
[Callback] Central Manager did update state to: Powered ON
Connecting to ENG004601...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to ENG004601
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Secure DFU...
Connected to ENG004601
Services discovered
Secure DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: FE59)
DFU characteristics discovered
Enabling indications for 8EC90004-F315-4F60-9FB8-838830DAEA50...
peripheral.setNotifyValue(true, for: 8EC90004-F315-4F60-9FB8-838830DAEA50)
Indications enabled for 8EC90004-F315-4F60-9FB8-838830DAEA50
Buttonless DFU indications enabled
Application with buttonless update found
Writing to characteristic 8EC90004-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x01, for: 8EC90004-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90004-F315-4F60-9FB8-838830DAEA50
Indication received from 8EC90004-F315-4F60-9FB8-838830DAEA50, value (0x):200101
Response (Op Code = 1, Status = 1) received
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
Buttonless service not configured, see: https://devzone.nordicsemi.com/f/nordic-q-a/59881/advertising-rename-feature-not-working/243566#243566. To workaround, disable alternative advertising name.
Connecting to ENG004601...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to ENG004601
Discovering services...
peripheral.discoverServices([FE59])
Services discovered
Secure DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: FE59)
DFU characteristics discovered
MTU set to 243
Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
Secure DFU Control Point notifications enabled
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000200000000000000000000
Command object info (Max size = 512, Offset = 0, CRC = 00000000) received
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x0101ec000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
Packet Receipt Notif disabled (Op Code = 2, Value = 0)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x080112a501a206a10108a8f0f4ea0710e89c031a028002200428a0ab0930d0f50242240803122089ab4613e58da4c2d2cdacea5604ba320d317972e2b0360cf08bef78ae8c9d3c524408031240a10aef5c149aaae8a73dbc2e0347d8ecd8c9e8e829086baaaecec3780029b0ef92490a331fca49ec637584eb6a131a2f837fb4389d40b9c7237b8c5482f8892c5200580178afea93ca01820110000000000000000000000000000000001a40724ea7245f91ef042fd92ab345423c797d99a35746268faa2344df43998c885944126d9aae43836ce9f1be7ea358ce8a13606a064358498a28577d7dafb8e0b9, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Command object sent (CRC = 4C25C37B)
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
[... 855 lines cut ...]
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
Data object 50/50 created
Uploading firmware...
Sending firmware to DFU Packet characteristic...
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030170100300141eccce
Checksum (Offset = 200816, CRC = CECC1E14) received
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
Data object executed
Upload completed in 30.39 seconds
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
Connecting to ENG004601...
centralManager.connect(peripheral, options: nil)
Connection timeout!
centralManager.cancelPeripheralConnection(peripheral)
Scanning for the DFU Bootloader...
centralManager.scanForPeripherals(withServices, [FE59]
Scanning timed out returning no matching peripherals!
centralManager.stopScan()

loves-bruins, 2021-03-29:
I found a fix for my problem.  Turns out the hard-coded timeout when connecting in DFUPeripheral.swift peripheralDidDisconnect() method wasn't waiting long enough.  I did make a branch, make the changes and tried to push, but I was having authentication issues.  So I created a patch file:
`diff --git a/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift b/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift
index 019f41b..e0a5e3a 100644
--- a/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift
+++ b/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift
@@ -654,7 +654,7 @@ internal class BaseCommonDFUPeripheral<TD : DFUPeripheralDelegate, TS : DFUServi
             if (delegate?.peripheralDidDisconnectAfterFirmwarePartSent() ?? false) {
                 // As we are already in bootloader mode, the peripheral set
                 // may be reused for sending a second part.
-                connectOrSwitchToNewPeripheral(after: 3.0)
+                connectOrSwitchToNewPeripheral(after: 15.0)
             } else {
                 // Upload is completed.
                 // Peripheral has been destroyed and state is now .completed.`
I don't really care on your approach.  It could be making this hardcoded 15, or even a user-settable property from service.

philips77, 2021-04-27:
We will release it in the next release.

closed 2021-04-27 by philips77

ManikandanRajamanickam12, 2022-08-29:
refresh() - device: xxxxxxxxx
I/DfuBaseService( 9486): Refreshing result: true
I/DfuBaseService( 9486): Cleaning up...
D/BluetoothGatt( 9486): cancelOpen() - device: xxxxxxxxxx
D/BluetoothGatt( 9486): close()
D/BluetoothGatt( 9486): unregisterApp() - mClientIf=8
I/DfuBaseService( 9486): Restarting the service
I/DfuBaseService( 9486): Starting DFU service in foreground
I/DfuBaseService( 9486): Connecting to the device...
D/BluetoothGatt( 9486): connect() - device: xxxxxxxx, auto: false
D/BluetoothGatt( 9486): registerApp()
D/BluetoothGatt( 9486): registerApp() - UUID=xxxxxxxxx
D/BluetoothGatt( 9486): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt( 9486): onClientConnectionState() - status=133 clientIf=8 device=xxxxxxxxxx
E/DfuBaseService( 9486): Connection state change error: 133 newState: 0
I/DfuBaseService( 9486): Connection error after: 30008 ms
E/DfuBaseService( 9486): Device not reachable. Check if the device with address xxxxxxxxx is in range, is advertising and is connectable
I/DfuBaseService( 9486): Attempt: 3
D/BluetoothGatt( 9486): refresh() - device: xxxxxxxx
I/DfuBaseService( 9486): Refreshing result: true
I/DfuBaseService( 9486): Cleaning up...
D/BluetoothGatt( 9486): cancelOpen() - device:xxxxxxxx
D/BluetoothGatt( 9486): close()
D/BluetoothGatt( 9486): unregisterApp() - mClientIf=8
I/DfuBaseService( 9486): DFU service destroyed
I/flutter ( 9486): PlatformException(133, DFU FAILED: GATT ERROR, Address: xxxxxxxxxxx, Error Type: 1, null)

I am facing this problem while upgrade a firmware.can any one help me to solve this problem....