nordicsemi/IOS-DFU-Library #369: Connection Timeout issue after Enabling DFU Mode

ibrar110, 2020-04-20:
Hi
I am facing a connection timeout issue. It seems after Enabling the DFU Mode, it fails to connect again. I tried using the nrfConnect app, the firmware seems to have no issues with it. I found ConnectionTimer in DFUPeripheral.swift class. Is there any way to increase the time from 10 Seconds to at least 30 seconds?
I am using the latest version of the DFULibrary, Xcode version is 11.1 and testing it on iOS 13.4.1.
Any help would be appreciated. Thanks in advance.

ibrar110, 2020-04-20:
Creating another connection timer really helped me out.

closed 2020-04-20 by ibrar110

philips77, 2020-07-31:
Option to set connection timer will be added in 4.8.0, release ongoing.

philips77, 2020-07-31:
Implemented in #374.